Ultimaker/Cura #13791: Automatically inserted horizontal fill layer is considered as part of outline (Printer prints into the air for the next outline layer)

danielroehrborn, 2022-11-11:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Anycubic Kobra Go

### Reproduction steps

I have loaded the STL file from blender into Cura, resized it a bit and run the slicer. Then I started the print.

### Actual results

The extra infill layer, that Cura decided to add in the middle of the model, did not include the outline. The printer was printing the first four of these horizontal infill layers but then it was moving too high on the Z-axis before it started the next outline layer. Because of this the outline layers were printed into the air. I assume the Z-offset must have been at least 2-3 layer heigths.

### Expected results

Cura should actually not generate these horizontal layers at all. But if it does then it should continue with the outline seamlessly and not cause the printer to move multiple layers up on the Z-Axis.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9991089/cura.log)
[CFFFP_Head.zip](https://github.com/Ultimaker/Cura/files/9991096/CFFFP_Head.zip)
[image]

danielroehrborn, 2022-11-11:
[image]
As I see now the additional layers are the top and bottom shell which are added because Cura has created a gap in the middle of the head. The STL model clearly does not have this gap.

danielroehrborn, 2022-11-11:
But why is this happening?

GregValiant, 2022-11-11:
Thanks for the report.
There are extra surfaces in the model and it is confusing Cura.
Go to the Cura MarketPlace and load the "Mesh Tools" plugin.  After loading you will need to quit Cura and restart it.
Re-load that model and then select the model and right-click on it.  The top menu item will be Mesh Tools.  Select "Check Mesh" from the dropdown.

Mesh Tools has some capability for repairing models.  MS 3D Builder is bundled with Windows and is more capable at repair.

Let us know how you make out.  This may be a Cura bug but we've seen that behavior before and it's always been an error in the model.

danielroehrborn, 2022-11-12:
I have noticed that when importing the model I get the warning that it is not watertight. The X-Ray view shows this:
[image]
When running the Mesh Check I have the following warning:
[image]
Because of this I tried to split the model into its pieces and remove all the unconnected ones. Then the slicer gives me this result:
[image]
It still wants to separate the top part of the head from the bottom, but this time even worse then before. Because now the upper parts of the ears are also cut of and the middle part of the head is replaced by supports. 
How do I have to change the model to not confuse Cura?

MariMakes, 2022-11-30:
Hey @danielroehrborn,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Your mesh seems to be faulty. 
I took a quick stab at repairing it, but you are losing quite some detail in the eyes.  [Head.zip](https://github.com/Ultimaker/Cura/files/10122272/Head.zip)
[image]

Best would be to return to the CAD software which was used to make it and make sure it's watertight. 

To me, it doesn't seem like a bug in Ultimaker Cura, since the mesh is faulty and Cura is not built to alter meshes in this way, that's what CAD software is for. 

Would you still like to keep this issue open or do you agree that we can close it?

danielroehrborn, 2022-12-02:
Thanks for your reply.
You can close the issue because I found a workaround:

1. I moved the eyebrows closer to the head until they touch the surface
2. Deleted all faces that are behind the eyebrows
Now Cura is not confused anymore and everything was printed very well :)