Ultimaker/Cura #13795: Slicing failed. Unexpected error.

magicsoccer, 2022-11-11:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Creality CR10 V2

### Reproduction steps

Open  cura.
Select object to print.
Slice

### Actual results

Slicing failed. Unexpected error.

### Expected results

Slicing without errors.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9992892/cura.log)
[image]
[spiral_planter_solidbottom.zip](https://github.com/Ultimaker/Cura/files/9992899/spiral_planter_solidbottom.zip)

GregValiant, 2022-11-12:
Thanks for the report.
I couldn't get your 3mf file to slice.  The model has some errors so I repaired it in MS 3D Builder and then tried again.
I still couldn't get the repaired model to slice with your settings.
I changed to the "Draft Quality" profile and changed the layer height to .25 and it sliced.  There may be some sort of conflict within your Cura settings.
Someone from the Cura team will take a look.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker