Ultimaker/Cura #13798: Slicing failed

shimmeryphil, 2022-11-11:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Creality ENder 3V2

### Reproduction steps

Tried to slice a file and it failed halfway through

### Actual results

I get a meesage saying "Slicing failed

### Expected results

Should have sliced

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9993675/cura.log)

File I was trying to print: https://drive.google.com/file/d/1cEFCn3ub3jVX6DO-YFFn0M-DMxoo__hC/view?usp=sharing

GregValiant, 2022-11-12:
Thanks for the report.
This is from your log file...
"Backend exited abnormally with return code 3221225477!"
There have been several slicing failures reported that show that error code.  I'll mark this as a duplicate and a slicing error.

shimmeryphil, 2022-11-14:
Thanks. By the way, I figured that switching to a different layer height
(.12 instead of .2) would force Cura to slice differently, possibly
avoiding whatever this bug is. It worked with slicing and printing
successfully completed.

Cheers
Phil

GregValiant, 2022-11-15:
I noticed that as well.  The "rotate or move" workaround can fool the bug as well.  We shouldn't need workarounds to get the program to slice when the model has no errors.

MariMakes, 2023-02-23:
Hey @shimmeryphil,

Return code 3221225477, is actually a super generic error, even though it doesn't sound generic 😅 
In the Cura 5.3 Beta, we made a fix that should solve a number of these errors. 
Can you download the Beta and retry your slice? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0-beta.1

If that's it still doesn't work, please let us know 🙏

closed 2023-03-10 by github-actions[bot]