Ultimaker/Cura #13800: Small objects aren't sliced correctly

Jan-NiklasB, 2022-11-12:
### Application Version

5.2.1

### Platform

Linux Mint 20.3, GTX 1070

### Printer

Custom Printer, Duet 3 based

### Reproduction steps

1. Created a model for a O-ring (7.8 mm ID, 8.6 mm OD, 1.1 mm H)
2. Imported the model to Cura
3. Set nozzel diameter to 0.2 mm
4. sliced

### Actual results

Only one wall line is generated (wall-width 0.8 mm, wall line width 0.2 mm)

### Expected results

4 wall lines are generated

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[Dichtung_Fokus_Laser.zip](https://github.com/Ultimaker/Cura/files/9994932/Dichtung_Fokus_Laser.zip)

GregValiant, 2022-11-12:
Thanks for the report.
8.6 OD is a radius of 4.3
7.8 ID is a radius of 3.9
4.3 - 3.9 = 0.4 cross section width.
Your "Outer Wall Inset" is 0.1.  Taking that from both the OD and the ID and you have a cross section that is .2mm across.  
Your Minimum Line Width is .34 which is wider than your Line Width.
Your Wall Width is .8 which is wider than the wall.
You can't hide the seam on the "Sharpest Corner" on a round object.

If your going to fool the program you need to cover all the bases.  Fix your settings and it will print with two loops instead of one.
I had to read the gcode into Auto-Cad as Cura wouldn't zoom in far enough.  There are two loops .2mm apart so that is the Line Width.
I'll go ahead and remove the bug label as this is a settings issue.

[image]

Jan-NiklasB, 2022-11-12:
Wow.... the 0.2 mm cross section thing was really a lack of overthinking the whole thing...but it even cost me a moment to figure out the correct setting (at least what the slicer shows it correctly to me now)...

Maybe those setting should be easier to find or set / more obvious, but don't ask me how to do this...

Anyways, thanks for your help!

closed 2022-11-12 by Jan-NiklasB