Ultimaker/Cura #13805: Support for older slicing engine

meltiseugen, 2022-11-12:
### Is your feature request related to a problem?

My printer does nou handle Arachne well

### Describe the solution you'd like

Be able to use cura 5 but with the old engine (like prusa slicer does)

### Describe alternatives you've considered

I have switched to prusa slicer because they support this option

### Affected users and/or printers

N/A

GregValiant, 2022-11-12:
[Cura 4.13.1](https://github.com/Ultimaker/Cura/releases/tag/4.13.1) was very good.  You can select the download from the list on the page I linked.  It was the last version before Arachne was rolled in.  All the 5.x versions use the variable line width introduced with Arachne.

meltiseugen, 2022-11-13:
> [Cura 4.13.1](https://github.com/Ultimaker/Cura/releases/tag/4.13.1) was very good. You can select the download from the list on the page I linked. It was the last version before Arachne was rolled in. All the 5.x versions use the variable line width introduced with Arachne.

Yes, but I also want to use the 5.x features (besides Arachne) without having to deal with variable line width.

Cura 4 will no longer have support or bug fixes ( like the seam bug)

MariMakes, 2022-12-06:
Hey @meltiseugen,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

We believe that the new Arachne Slicer Engine is the best way forward and cannot start supporting two slicing engines in tandem without losing focus on the most requested improvements from our community and from Ultimaker. 

If you are running into trouble using the new engine on your machine I would like to refer you to this support page where you can learn more about the different settings you can use to finetune the settings for your printer. There you can also find the settings that you can alter to limit the amount of variation in your lines. https://support.ultimaker.com/hc/en-us/articles/4792077687068

I'll be closing this issue, I hope you understand.

closed 2022-12-06 by MariMakes