Ultimaker/Cura #13807: Slicing failed with tree support on mask - Sovol SV06

BigRigCookin, 2022-11-13:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Sovol SV06

### Reproduction steps

tried to put tree support on a mask im printing

### Actual results

Wouldnt slice

### Expected results

slice iwth tree supports

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-11-13:
Thanks for the report.
Without a Cura.log file ("Help | Show Configuration Folder") and Project File ("File | Save Project") there can be no investigation.
Github requires that the files be zipped and the zip folder can be posted here.

GregValiant, 2022-11-14:
@FlyingWolf11 I sliced your file and didn't have any trouble so I am unable to reproduce the problem.  I'm on Win10.
[image]

This appears 6 times in your log file:
"Backend exited abnormally with return code -11!"
Someone from the Cura team will take a look.

@BigRigCookin We could use a file from you to see if this is the same sort of error.

FlyingWolf11, 2022-11-15:
@GregValiant I've tested something out and it seems like the slice with tree supports only fails if the model is rotated specifically. In my case the file I gave you yesterday still doesn't work but it does when it is rotated by -90 degrees on the x axis.

GregValiant, 2022-11-15:
If the model slices when rotated then that is an often reported bug.
When it gets fixed there will be a lot of reports cleared up.

FlyingWolf11, 2022-11-15:
That is good to know. I was thinking the tree supports could be the fault since the slicing worked with rotation and normal supports but not with rotation and tree supports.

GregValiant, 2022-11-15:
No, it's a slicing bug as far as I know.  Mac and Linux don't seem to be affected, just Windows systems.

closed 2022-11-27 by github-actions[bot]