Ultimaker/Cura #13808: Unable to slice model

Yukun99, 2022-11-13:
### Application Version

5.2.1

### Platform

Windoows 11, RTX 3080

### Printer

Flsun Super Racer

### Reproduction steps

1. Open up model
2. Slice model

### Actual results

Slicing errors out.

### Expected results

Model should be sliced.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

https://www.thingiverse.com/thing:5562934/files

Yukun99, 2022-11-13:
It appears the issue is with my slicer settings, here is my settings profile:
https://drive.google.com/file/d/1vEmXIppP9iY5I-W5_Ustnaoxfwu5D90i/view?usp=sharing

GregValiant, 2022-11-13:
Thanks for the report.
There may be a Cura bug here but look at this view of the model in Mesh Mixer.  Each of those magenta and red balls indicate an error in the model.  All of the red colored surfaces have errors.
[image]

Here is the report from OnllineSTLRepair:
--> 709 Naked edges (?)
--> 13 Planar holes (?)
--> 27 Non-planar holes (?)
--> 2 Non-manifold edges (?)
--> 344 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 19516ms ------
-> Vertex count changed from 8866 to 13648 (+4782)
-> Triangle count changed from 17429 to 27794 (+10365)

This isn't as bad as the "Crystal Dragon" (which is also an articulated dragon) but this is pretty bad.
MS 3D Builder was able to repair it as was the OnLineSTLRepair site.
I had no problem slicing the repaired model.
You may want to inform the designer of this that there are serious issues with the model.
I'm going to remove the bug label as this is a model problem.  I hope you understand.

[image]

Yukun99, 2022-11-13:
> Thanks for the report. There may be a Cura bug here but look at this view of the model in Mesh Mixer. Each of those magenta and red balls indicate an error in the model. All of the red colored [...]

Ah, I did not receive an error on the model being non manifold etc. like I normally do with bad models so I thought it was fine, thanks for the info!

closed 2022-11-13 by Yukun99