Ultimaker/Cura #13809: (5.2.1)Cura program crashes on loading

KubaMed, 2022-11-13:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Longer 5 pro

### Reproduction steps

1. Clicked on Ultimaker Cura 5.2.1.exe 
2. cura stars loading

### Actual results

3. crash before actual cura window opens (sometimes I get on screen in screenshot1, but then it crashes)

error messages I got are from windows -> Ultimaker-Cura.exe stopped working 
->
when I close this error I gen another windows error message
->
Slicing engine for Cure stopped working 

### Expected results

Cura loads

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

This was fresh install, I deleted everything, restarted pc and then installed Cura 5.2.1
I tried to install older random versions from cura 4 to cura 5 -> no luck, same crashing issues 
Cura was crashing on me few weeks already (I was using old version on my laptop that is stable)

I dont know what old version of cura it was, but after fresh install everything seems to be wokring, but after I added the Longer LK5 pro Printer from preset It crashed) 

Windows 10 version - 10.0.19043 Build 19043
czech language
PC specs:
2 monitors
GPU: NVIDIA GeForce RTX 3060
GPU driver: geforce game ready driver 526.86
GLSL Version:    4.0.0
OpenGL Version:  4.1.0 NVIDIA 526.86
CPU: Intel(R) Core(TM) i5-6500 CPU @ 3.20GHz, 3201 Mhz, cores: 4, threads: 4
RAM: 16GB ddr4

If you need more speciffic information prase tag me I will be chekcing this issue.

LOG HERE 
[cura.log](https://github.com/Ultimaker/Cura/files/9997004/cura.log)
[image]

MariMakes, 2022-12-08:
Hey @KubaMed,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I have a couple of questions to narrow it down.
	1. Do you have multiple monitors? Does Cura start if you disable one of the monitors?
	2. Do you have GPU Tweak OSD overlay enabled? Does Cura start if you disable that software?
	3. Do you have a Ryzen Graphics card? Can you try this workaround? https://support.makerbot.com/s/article/1667412437627

If this doesn't help, let me know and I'll bring it up to the team to take a look. 😉 
Good luck 💪

KubaMed, 2022-12-08:
Hi @MariMakes 
Thank you so much! 2. Disabled OSD overlay fixed this issue, thank you so much again! 🙇

KubaMed, 2022-12-08:
GPU Tweak OSD overlay issue

MariMakes, 2022-12-08:
Huzzah! 🎉 So good to hear!

Happy printing!

closed 2022-12-08 by MariMakes