Ultimaker/Cura #13810: Inserting objects

Chris3126, 2022-11-13:
### Is your feature request related to a problem?

I don't have a problem, just a suggestion for an improvement.

### Describe the solution you'd like

I frequently use Cura and often I insert objects, such as nuts, magnets etc. I wonder if it is possible to add a button with the aim to insert a print pause in the G-code after scrolling up/down through the printsteps to select the layer where to pause. That way it is not neccesary to watch the printer all the time.

### Describe alternatives you've considered

Right now the only way to insert objects is to watch the printer and manually start the pause and insert the objects.

### Affected users and/or printers

This will be quite benneficial for many users and make the program even more competitive.

GregValiant, 2022-11-13:
The Pause at Height post-processer is exactly for this (and changing filament color).
Go to "Extensions / Post Processing / Modify Gcode" and then "Select Script" and choose "Pause at Height" from the list.
"By Height" doesn't work well if there are Z-hops in the file or if you are using "Adaptive Layer Height" but "By Layer" works very well.
Fill out the choices in Pause at Height.  The layer number you enter will be the last layer printed before the pause.  If you are inserting a nut other object you will want to park the head at the rear so enter your Max X and Max Y as the Park position.
In "Gcode before Pause" enter M300 and you will get a beep when the pause occurs.
The max allowable "Disarm Timeout" time is 1800 seconds (30 minutes) so enter 1800.  If the stepper motors disarm they lose their position and the print will basically be scrap at that point.
Make the "Standby Temperature" your "Print temperature".
M0 is the standard pause command for a lot of printers but some firmware doesn't support it.  M25 is an alternative.
If you use M0 then clicking the LCD button will resume the print.  I grab the string from the nozzle and hold it then click the button and when the print head moves back to the print position I snap my wrist and break off the string.  It works well.
When changing colors I don't bother with the "retraction" settings and instead just give the filament a manual retaction of about 5mm.  That works for my printer.
It does require some practice but the restart can be made totally seamless.
Here I have inserted two nuts in a print.  I had the head park at the left rear.
[image]

Chris3126, 2022-11-13:
Thanks I will try this option. But I also hope that the developers can introduce an easier solution.

closed 2022-11-13 by Chris3126