Ultimaker/Cura #13812: random lines in parts from cura, did not do it in previous versions

MarlinMerlin, 2022-11-13:
### Application Version

5.2.1

### Platform

windows 10

### Printer

cr 10 s5

### Reproduction steps

Slice a part with any profile

### Actual results

there are lines visible where it should be smooth. These show up in the actual print as well. 

### Expected results

The prints used to be nicer and smoother

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

GregValiant, 2022-11-13:
Thanks for the report.
From here that looks like something going on in the model.  The white points might be the Z seam.
Without a Project File ("File | Save Project") that's all the information you have provided us to go on.

MarlinMerlin, 2022-11-13:
@GregValiant Thanks for the quick reply. I updated the original post to include more information, images, and a project file. Let me know if you're able to view the zipped project and what else I can provide to help!

GregValiant, 2022-11-14:
I checked the model in MS 3D Builder and it said there were errors and was able to fix them.  I also checked with OnLineSTLRepair but it didn't find any errors.  I ended up slicing both your model and the repaired model and I didn't see any difference between the slices.

The bottom of the model isn't flat and so the model doesn't sit flush on the build plate.  That is creating problems in PrusaSlicer which is declaring it to be "unprintable".  When I slice the model in Cura using your settings there is a definite oddity starting at layer 140 (49mm).  Some of the extrusion lines are set back from the preceding layer.  That's what is causing the lines you see in the preview.  Reading the gcode into AutoCad the set back is about 0.11mm.

Looking at the Prepare screen the model looks great and slicing it in both Surface mode and in Spiralize the "setback" oddity doesn't show up.  The front surface looks good in both modes.
In MS 3D Builder the front surface show as a smooth surface.  (The red lines around the model is the error indicator as I hadn't repaired it yet.)
[image]

I'm in "compatibility" mode on this computer.  It's simpler but it has certain advantages in the previews.
In the image below you can just make out that the top two layers (140 and 141) don't quite line up with 139.  That's causing the line in your preview that you reported.  But Cura has variable line width and may be compensating so the width of the extrusion does line up with the previous layers in the print.  I'm tending to think that (but I did not print the model).

The results from my Surface Mode test and Spiralize test seem to indicate that Cura is seeing the model correctly.  BUT the backside wall doesn't show this "oddity".  It would seem logical that if Cura was doing something with the flow and the line width was increasing then there would be similar areas on the backside showing the same behavior.  I did not see any evidence of that.
Looking at the line widths that Cura is using there is a lot of .53 line width which appears to relate to the support and a then a lot of .70 to .78 line widths which are pretty much everything else.

I guess I have one question and that is:  If you printed this did those lines show up as defects in the print?  if they did then this is likely a slicing issue.  If they didn't then it may be a Previewing issue.  
[image]

MarlinMerlin, 2022-11-14:
Thank you for the significant time and help on this issue, @GregValiant. These artifacts are present on the final part in identical locations after it has printed. Orientation and model seem to not matter as it's occurred on several different prints for me now.

I uploaded a zip in this comment which should have a flattened base project version, which still exhibits the anomaly. The back side **does** indeed have the layer issues as well (attached). On the front face, layer 'shifting' appears around layer 26, 102, 109, 136, 138, 150, 152. On the rear, 18, 103, then several times from 116-130.

[project flattened.zip](https://github.com/Ultimaker/Cura/files/10003411/project.flattened.zip)

GregValiant, 2022-11-14:
In the new model there is a significant event from Layer 12 to Layer 13.  It appears that the line width of both the outer walls and the inner walls is too narrow on layer 12(?)  At any rate there is a difference in line width on both the front and rear walls there.
I think this is enough info to get the Cura team a start when they look at this.

Thanks for the files.  So often people don't provide any information and then expect an immediate fix.

MarlinMerlin, 2022-11-15:
Hopefully the team can take a look and remedy the issue since 5.x has many improved features I like to use! Thanks for the time and help - let me know if there's any other info I can provide to give cura devs a fighting chance.

MarlinMerlin, 2023-07-06:
Hi guys, back from the dead. Just checked out 5.4.0 and it's still present. I've figured out the root cause of the issue, and it has to do with the adaptive line widths (arachne). The line widths will vary slightly, and in a big flat face like that it's super noticeable. Is there a way to disable this somewhere in the settings? I've tried the easy fix of setting min line widths to the primary extrusion width but it still gives me those weird wiggles.

GregValiant, 2023-07-06:
This doesn't appear to have been fixed in 5.4.0.
[image]

At the starts and ends of those lines there is no inset.  The problem I see is that the middle portion doesn't seem to follow the model.  
[image]