Ultimaker/Cura #13813: Slicing error

R-B-Riddick, 2022-11-14:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

1. Opened a file (Head of son goku dragonball)
2. Changed Tree Support from touching buildplate to everywhere

If i change back to touching its working again. 
the slicing takes a relatively long time and the printing time also seems longer than normal to me. 

### Actual results

slicing had an error

### Expected results

that the slice should work

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/10000621/cura.log)

[CE3PRO_Cabeza 01.zip](https://github.com/Ultimaker/Cura/files/10000604/CE3PRO_Cabeza.01.zip)

GregValiant, 2022-11-16:
Thanks for the report.
Your project sliced fine for me so I am unable to duplicate the problem.

There have been several reports about a slicing bug that can sometimes be fooled by rotating the model on the build plate.  Give that a try and let us know if it works.
I'll leave this open and as a bug report for now.
[image]

R-B-Riddick, 2022-11-16:
You are right. I just opened it again and sliced it without moving and it worked. Guess there was something else in my profile. I'm sorry that you had to do more work unnecessarily

closed 2022-11-16 by R-B-Riddick