Ultimaker/Cura #13815: New PLA support with periodic dissolvable layers (i.e PLA + PVA)

ok4global, 2022-11-14:
### Is your feature request related to a problem?

PVA and other dissolvable are finicky and expensive and not as reliable as PLA, ABS etc.

PVA is finicky, expensive, and not as reliable as printing with PLA, so I print my supports in PLA and the interface layer in PVA.   
It works, but it is still hard to get the PLA supports off.  It would be fantastic if the PLA support could have periodic layers of PVA so that the PLA support structure would fall apart once the PVA dissolves. 

### Describe the solution you'd like

New support profile — PLA support with periodic layers of dissolving support like PVA

For example: 
Model =  extruder 1
Support: 
      10 layers of PLA (extruder 1)
      1 layer of PVA  (extruder 2)
      10 layers of PLA (extruder 1)
      1 layer of  PVA  (extruder 2)

For stability, the PVA layer could be 95% PVA with dots of PLA.  So the PVA layer could look like:

PVA————— •PLA•—————PVA————— •PLA•—————PVA

### Describe alternatives you've considered

I print my supports in PLA and the interface layer in PVA.  It works, but it is still hard to get the PLA supports off. 

### Affected users and/or printers

Everyone!

MariMakes, 2022-12-06:
Hey @ok4global,

Thanks for your report 👍

Have you considered using Break Up Support In Chunks? 
It sounds super similar, but instead of adding a layer of PVA the layer is skipped, so it can be broken off easily. 
PLA•—————Empty————— •PLA•
[image]

What do you think? Will this also solve your problem?

ok4global, 2022-12-06:
Hi,

Thanks for writing back!

Break UP Support in Chunks looks great and I will try it!!!!  Thank you!

I do think adding an intermittent PVA option would leverage the Ulitmaker’s 2 nozzle extrusion.  Also, I think having the PVA nozzle extrude something on every layer would keep the PVA primed and working better.

Just an idea, but could also be like this:

• PVA
—PLA

—————•••••••••••••—————
——•••••••••••••————————
•••••••••••••——————————
••••••••——————————•••••
•••——————————••••••••••
—————————•••••••••••••—
——————•••••••••••••————
———•••••••••••••———————

Thanks!  Steve