Ultimaker/Cura #13822: Cura 5.2.1 Marketplace causes Crash

HansiSolo, 2022-11-14:
### Application Version

5.2.1

### Platform

Windows 11 Pro 22h2, Ryzen 3950x, GTX1080

### Printer

Creality Ender 3

### Reproduction steps

Running Windows 11 pro build 22h2, and using Cura 5.2.1 anytime I open the marketplace the entire app becomes nonresponsive and crashes.

### Actual results

After it closes and I try to reopen I get a blank screen and is still unresponsive. I found through the Ultimaker forum to delete the the appdata local and roaming cura folders, and that fixes that problem.

### Expected results

Be able to get into marketplace

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Cura Log Files attached
[cura.log](https://github.com/Ultimaker/Cura/files/10007027/cura.log)

sgessner, 2022-11-28:
I have the same Issue.
Could it be because I don't have an Ultimaker account and the Marketplace can't open without logging in?

fieldOfView, 2022-11-28:
> Could it be because I don't have an Ultimaker account and the Marketplace can't open without logging in?

No. I am never logged in to my Ultimaker account. The Marketplace should work fine without being logged in.

Do you have multiple displays? On some systems, Cura has trouble opening dialogs on the non-primary display, and on some systems, dialogs open on arbitrary displays.

MariMakes, 2022-11-30:
Hey @HansiSolo,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Thanks for your detailed report 👍
I'll have someone from the team look at your logs. 

In the meantime, are you looking for a specific plugin? 
Or did removing the folders get you unstuck?

HansiSolo, 2022-12-02:
sorry for the late response:

I was trying to get into the plug-ins to see if someone had already put filament settings for the filament I’m currently using. Removing the folders only gets me unstuck until I try to re-access plug-ins or the marketplace, which I’ve done a couple time accidentally. After I delete them then I have to go back in and setup my printer settings all over again.
I do have multiple monitors, but I’ve tried doing things on my secondary and primary screen with same result.

MariMakes, 2022-12-09:
Hey,

We see a lot of requests being made as if Cura thinks you were logged in.
But since you weren't Cura kept asking for account information and crashed. 
By removing your folders Cura forgot that you were signed in before which made the crash go away.

If you encounter it again, just removing this cura.cfg should be enough, so you can keep your profiles and printers. 
[image]

We'll keep an eye out to see if we receive similar reports.
Thanks again. 💪