Ultimaker/Cura #13823: Sign in failure to Ultimaker cura software

ChamikaNir, 2022-11-15:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ultimaker S5 and Ultimaker S3

### Reproduction steps

Hi,
These are the steps I followed,

1. click on the Sign-in button in cura software 
2. Chrome opens and asks me to log in
3. login was successful and I was redirected to localhost
4. Then I got "ERR_EMPTY_RESPONSE" as below
[image]

The localhost address was http://localhost:32118/callback?code=y1iiyhUPGseYTwnoskgGbVITDTzbgQtH&state=7f03a72575e5c025521ff8af5d2fb15a9fd95eaa8557b542779c0fd121c1daa0

### Actual results

I can't log in to CURA software and print via the cloud.

### Expected results

Log in to CURA software and check available printers and print via the cloud

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10010037/cura.log)

ChamikaNir, 2022-11-29:
Hi, Any progress on the above issue?

MariMakes, 2022-11-30:
Hey @ChamikaNir,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

This is the first time I've seen a report like this.
I'll have someone from the team take a look at this.

Could you install this extensive logging plug-in?
https://marketplace.ultimaker.com/app/cura/plugins/UltimakerPackages/ExtensiveSupportLogging
 
If it's loaded could you try to sign in, export the zip, disable the plug-in and share the results with us? 
That will help with troubleshooting. 

Thank you 😇

ChamikaNir, 2022-12-01:
Hi @MariMakes ,

Thank you very much for your response.
Please refer the below file for your further troubleshooting. 
[remote-support_22-12-01-13_06_40.zip](https://github.com/Ultimaker/Cura/files/10130260/remote-support_22-12-01-13_06_40.zip)
 
Also I have tried followings as well,
1. reinstall CURA 5.2.1
2. Install the previous version (CURA 5.1.0)
3. delete localhost from hsts as suggested in the https://github.com/Ultimaker/Cura/issues/9772
4. Reinstall windows 11.

But non of the above solve the problem. 
Please do reach me if you need any further information. 

Thank you.

MariMakes, 2022-12-09:
Hey @ChamikaNir,

I brought your issue up to the team.
We were a bit shocked to see that you had to re-install Windows 11. That seems like a lot of work. 😮 

They think it might be related to your firewall. 
Do you have a special strict company firewall that cause this?

We have a special support page that might help you with whitelisting in this case.  
https://support.makerbot.com/s/article/1667337958118

Good luck. 💪  Can you let us know if this helps?

ChamikaNir, 2022-12-09:
Hi @MariMakes,

Yes, We have special strict firewall. 

But my other colleges who have same firewall restrictions can and login and use cure without any problem. I was using the previous versions (CURA 5.1.0) and other older versions without any issue for longer period of time. 
This specific issue came up once I updated to the 5.2.1. 

I already checked with our IT support team on ports and domain. All the ports mentioned above are open and all the requested domains are whitelisted. 

I hope this is not related to firewall.

ChamikaNir, 2023-01-05:
Hi @MariMakes,
Hny update?

MariMakes, 2023-01-05:
Hey @ChamikaNir,

So sorry I didn't get back to you, I lost track of the ticket. 
I'll have to bring it up again with the team again.