Ultimaker/Cura #13824: Error Signing the Bundle [Build from source]

julianCast, 2022-11-15:
### Application Version

5.3

### Platform

Mac OS Monterey

### Printer

No apply

### Reproduction steps

I managed to create the bundle but the signing step failed. 
```shell
conan install . --build=missing --update -g VirtualPythonEnv -o cura:devtools=True --json "venv/conan_install_info.json" -if cura_inst
source venv/bin/activate
pyinstaller venv/conan/Ultimaker-Cura.spec
```

### Actual results

```shell
417526 INFO: Signing the BUNDLE...
417916 WARNING: Error while signing the bundle: codesign command (['codesign', '-s', 'E7433F07900F4A61972DCA607EE11ADD774B65EC', '--force', '--all-architectures', '--timestamp', '--options=runtime', '--entitlements', '/Users/julian/Cura5/packaging/dmg/cura.entitlements', '--deep', '/Users/julian/Cura5/dist/Ultimaker-Cura.app']) failed with error code 1!
output: /Users/julian/Cura5/dist/Ultimaker-Cura.app: replacing existing signature
/Users/julian/Cura5/dist/Ultimaker-Cura.app: .DS_Store files cannot be a symlink

417916 WARNING: You will need to sign the bundle manually!
417916 INFO: Building BUNDLE UMBUNDLE-00.toc completed successfully.
```

### Expected results

```shell
408776 INFO: Signing the BUNDLE...
495580 INFO: Building BUNDLE UMBUNDLE-00.toc completed successfully.
```

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I previously imported my certificates into keychain and installed the utils (autoconf automake ninja create-dmg).

MariMakes, 2022-11-30:
Hey @julianCast 

Sorry, it took us a while to get back to you 😞 

I don't know a lot about building, but I do hear that signing builds is super tricky.
I'll bring it up to the team to see how we can help you best. 

I'm super curious. What do you want to achieve by signing a Build?
Are you intending to share it with others or are you using it for research purposes?
I would love to know more about your project, maybe we can help.

casperlamboo, 2022-11-30:
Hi @julianCast 

Did you set the `MAC_NOTARIZE_USER`, `MAC_NOTARIZE_PASS` and `CODESIGN_IDENTITY` environment variables?  These are needed to sign/notarize the app.

The user key is usually the email adress you use to login to the apple developers portal. The `MAC_NOTARIZE_PASS` is _not_ the accounts password but an app specific password that can be generated from the apple developers portal.

If you have installed the correct certificates in your key-chain then the `CODESIGN_IDENTITY` can be found using the following [command](https://stackoverflow.com/questions/16576537/how-can-i-find-exactly-what-my-codesign-identity-is)
```
$ security find-identity -v -p codesigning
```

The environment variable `CODESIGN_IDENTITY` should be the XXXXXX part in the result
```
$ security find-identity -v -p codesigning
> 1) XXXXXX "Developer ID Application: SomeCompany (SOME_ID)"
```

casperlamboo, 2022-11-30:
Note that this step is not needed for development, this is just needed to generate a `.dmg` for distribution. If you want a build for yourself or to use internally the unsigned `.app` file works aswell.

julianCast, 2022-12-13:
Thank you! I did was using the `MAC_NOTARIZE_USER, MAC_NOTARIZE_PASS, CODESIGN_IDENTITY` but the certificate was not correctly installed in the system.

closed 2022-12-13 by julianCast