Ultimaker/Cura #13826: 5.2.1 doesn't update G1 feedrate after initial layer, incorrectly assuming G1 feedrate equals G0 one.

Nikolay-Po, 2022-11-15:
### Application Version

5.2.1

### Platform

Debian 11

### Printer

Customized, Marlin bugfix-2.1.x

### Reproduction steps

Slice single wall model with different speed setup for initial layer and for walls.
Print resulting g-code.

### Actual results

When I'm printing the g-code, the speed of wall printing on second and subsequent layers remains the same as at first layer.
The slicer doesn't put G1 F parameter when it needed at start of second layer.
With different line types the problem is mitigated because the speed is updating naturally between different G1 moves and the problem is unnoticeable because persist only until first G1 F command feerdate change.
But mine case is specific. I have only one type of extrusion motion, so, none switching between different G1 feedrates is used.
Project and resulting g-code are in attached ZIP file.
[Input_Shaping_Test_09_Off.zip](https://github.com/Ultimaker/Cura/files/10011930/Input_Shaping_Test_09_Off.zip)

### Expected results

Feedrate update for G1 extrusion moves is expected when starting second layer.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

The problem is here, see 5641 line of attached g-code:
```
                                       ;This is the end of first layer (LAYER:0) at 60mm/s wall speed
                                       ;See line 5635 in attached g-code
G1 X75.2 Y225.007 E1942.61384
G0 F11280 X74.978 Y224.786
G1 F3600 X75.2 Y225.007 E1942.63452    ;G1 feedrate was set last time here. It is 60mm/s.
M104 S250
G1 X74.978 Y225.012 E1942.64917
G10
;MESH:NONMESH
G0 F600 X74.978 Y225.012 Z0.3          ;Lifting a nozzle to second layer (LAYER:1).
G0 F11280 X144.659 Y144.548            ;Traveling at correct speed of 188mm/s. G0 speed updated to 188mm/s but G1 speed is not!
;TIME_ELAPSED:562.745528
;LAYER:1
;TYPE:WALL-OUTER
;MESH:Input_Shaping_Test_05_Off.3mf
G11
G1 X144.537 Y144.665 E1942.65475       ;Printing second layer and further at slow, 60mm/s speed because there were none G1 F11280 command.
G1 X79.455 Y144.659 E1944.80246
```
Current Cura assumptions of G0 and G1 speeds are different from Marlin Firmware machine. When I'm opening the g-code in Cura itself, all seems to be correct, take a look at a screenshot: second and above layers speed is 188mm/s by setup.

[image]

But printer firmware assumption is different. Firmware treats G0 feedrate value different than G1. It seems to me correct in sight of common CNC software. Take a look at a discussion at [duet3d.com](https://forum.duet3d.com/post/42808), a dialogue:

>- When machining both G0 and G1 are extensively used. G0 is normally used "when not cutting" (as specified by NIST) ... Normally G0 feedrate is machine specific, if not specified otherwise.
>- Are you saying that the F parameter should only apply to G1 moves, and G0 moves should always execute at the machine limits set in M203?
>- Almost! G1 should work as it works right now. G0 should use M203 limits unless Fnnnn is specified with the code.

So, I'm expecting Cura slicer will generate G0 for travel motion and G1 for printing (or retracting, etc) and will keep feedrates separate and appropriate for both commands.
G0 feedrate shouldn't assumed to be G1 feedrate. Dedicated G1 feedrate setup by G1 command with "F" parameter should be made if printing speed is changing.

GregValiant, 2022-11-15:
Thanks for the report.
I can duplicate this with your project file.  After poking around with the settings it appears that two settings are interacting badly or possibly it's a conflict with something in your definition file.

For your printer and setting profile - if "Firmware Retraction" is enabled then "Minimum Layer Time" **_must_** be 10 seconds or greater.

Switching to my Ender and a different profile, I could not duplicate the problem.

The Cura team will take a look.

Nikolay-Po, 2022-11-16:
Mention must be noted that (IMO) Cura incorrectly interprets flowrate settings when importing gcode file. An example is g-code in attached archive above. Last G1 speed was 60mm/s but Cura exhibits 188mm/s. That is why I thought it uses one flowrate value for both type of motion, irrespective which was changed last.
I sure it will be correct to interpret G0 and G1 speeds separately. Thank you for attention to the problem.

GregValiant, 2022-11-16:
My printer (Creality with Marlin) uses the same feed rate for G1 and G0.  Change one and the other changes.  There does not appear to be a "G1FeedRate" and a separate "G0FeedRate" but rather it's "FeedRate".
So in this snippet from your file:
G0 F11280 X144.659 Y144.548
;TIME_ELAPSED:562.745528
;LAYER:1
;TYPE:WALL-OUTER
;MESH:Input_Shaping_Test_05_Off.3mf
G11
G1 X144.537 Y144.665 E1942.65475
On my printer that extrusion will be at 188mm/sec as that was the last change to what I am calling "FeedRate".

Nikolay-Po, 2022-11-16:
@GregValiant, thank you for citing Marlin G0/G1 behavior description. Indeed, in mine configuration I'm using custom G0 feedrate as homing speed and bed leveling traveling speed. Also, to get some flexibility for testing purposes I allowed G0 to be variable, may be the last is a cause of different g-code parser behavior:
```
// Enable and set a (default) feedrate for all G0 moves
#define G0_FEEDRATE 3600 // (mm/min)
#ifdef G0_FEEDRATE
  #define VARIABLE_G0_FEEDRATE // The G0 feedrate is set by F in G0 motion mode
#endif

```
So, mine usage assumptions are partially in a conflict with Marlin usage directions. I should issue a feature request instead of a bug report.
In sight of view of current Marling Firmware documentation, there are none bugs in Cura's G0/G1 usage. So, I'm closing this issue.

closed 2022-11-16 by Nikolay-Po

GregValiant, 2022-11-16:
Thank you.
This is something that may need to be addressed at some point as firmware moves on.  Bringing it up now may be helpful to get people to think about it.

Nikolay-Po, 2022-11-17:
I checked the case better and found some my misunderstandings. Now #13840 is better than this ineffective "bug report".

GregValiant, 2022-11-17:
I think so too.  It may get some additional attention as a Feature Request as well.