Ultimaker/Cura #13829: object of type SettingFunction is not JSON serializable

Linkenelis, 2022-11-15:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

modified Saphire plus

### Reproduction steps

Print with octoprint

### Actual results

error: object of type SettingFunction is not JSON serializable

### Expected results

Should have written the file to Octoprint just like 5.1.0 still does

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Error:   [image]
I still have the previous version, that keeps working as normal

fieldOfView, 2022-11-27:
Neither @Kedryn nor @Linkenelis have provided logs. That makes it harder for us to fix this.

Please see https://github.com/Ultimaker/Cura/blob/main/docs/Report.md#cura-gui-log about where to find the logs and post them here.

fieldOfView, 2022-11-27:
This is also a duplicate of #13675, so I am closing this issue in order to keep the discussion centralized.

closed 2022-11-27 by fieldOfView

Linkenelis, 2022-11-27:
@fieldOfView : sorry for not sending a log...
Update is that it started again, for no reason that I can see. It might be the reboot of my PC that fixed it.