Ultimaker/Cura #13833: Search and replace modifies how Cura generates de gcode

Jrribas, 2022-11-15:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Trident 300

### Reproduction steps

1. Generate a gcode without any post processing scripts. 
2. Save to a file
3. Add Search and replace to the post processing scripts.
4. Add ";Generated with Cura_SteamEngine 5.2.1" to first field and "1" to second, regexp turned off. This also works without inserting anything.
5. Save to a diferent file.
6. Compare both with kdiff or look trough the first layer. The first difference is the 4th wall of the outer wall.

### Actual results

The files are diferent not only with respect to the search and replace functionality. 
Basically the gcode without the search and replace, in my case, for one wall of the cube, CURA would always make two commands to complete the wall. One from the corner to the middle of the wall and another from the middle to the other corner.
With the search and replace enabled CURA only generates one command. For the same wall the command goes from one corner to the other.

### Expected results

The gcode itself should not change apart from the strings that need to be replaced.
I even think the gcode with search and replace enabled is better as it saves steps, but this changes should not happen.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[bug.zip](https://github.com/Ultimaker/Cura/files/10015811/bug.zip)

GregValiant, 2022-11-16:
Thanks for the report.
Search and Replace simply finds what you told it to look for and then inserts something else.  If it doesn't find the "Search Criteria" it does nothing.  There is nothing else in it's code that it can do.
Someone from the Cura team will respond but I'm pretty sure the changes in the gcode are due to variations and rounding off in the math algorithms as they go through the data that a model represents.

I believe you will find that if you slice a file 50 times it will have subtle differences 50 times.
I disabled Search and Replace and sliced the cube 5 times.  To get Cura to re-slice I backspaced the "2" from the layer height and then typed it in again.  I did that 5 times.
Here are the differences in file size in bytes:
330115
330112
329568
329469
329917
330238 is your "No Post Processing" file.

This is the way the program works.  I don't see a bug here but I'll leave the label on for now.

Jrribas, 2022-11-16:
Thank you for your answer. Yes, I just tested by making several exports always with post processing off and it seems to be the case. I never expected that CURA with the same settings and same model in the same position would have diferent gcodes outputs. The more you know xD

fieldOfView, 2022-11-16:
The slicing process should be deterministic though. Same input -> same output. Otherwise it would not be possible to do any automated testing on CuraEngine.

GregValiant, 2022-11-16:
I tried a line-by-line comparison but it didn't work for me because the ";LAYER:" comment appears on different lines within each file.  I thought the line number where LAYER shows up would make a good comparison.

This shows the line number within the gcode file where ";LAYER:"  appears within each file:
...............Slice: 1....................................Slice: 2....................................Slice: 3....................................Slice: 4......................................Slice: 5
;LAYER:25 @ 2531................;LAYER:25 @ 2520..............;LAYER:25 @ 2524..............;LAYER:25 @ 2527..............;LAYER:25 @ 2521
;LAYER:50 @ 4350................;LAYER:50 @ 4348..............;LAYER:50 @ 4349..............;LAYER:50 @ 4342..............;LAYER:50 @ 4349
;LAYER:100 @ 7991............;LAYER:100 @ 7982...........;LAYER:100 @ 7971...........;LAYER:100 @ 7962............;LAYER:100 @ 7989
;LAYER:149 @ 12020.........;LAYER:149 @ 12020.........;LAYER:149 @ 11998.........;LAYER:149 @ 11994.........;LAYER:149 @ 12012
LastLine..... @ 12248...............LastLine @ 12248...............LastLine @ 12222...............LastLine @ 12222...............LastLine @ 12240

As indicated by the file size - @Jrribas slice was the longest file at 12253 lines.

Another slice this morning produced yet a different result.

So something appears to be varying from slice to slice.  Looking at Layer 25 of slice 1 and slice 2 there is a  -11 line difference but by layer 149 it has swung back to no difference.

fieldOfView, 2022-11-16:
I think @rburema should be able to tell us if CuraEngine should be deterministic, and whether it is a problem if it isn't.

rburema, 2022-11-16:
While I can see the problem with non-determinism in CuraEngine (just for starters, it can foul up debugging as well), at the moment I don't think we (can) expect CuraEngine to be completely deterministic.

Just as an example, we use `std::sort(...)` instead of `std::stable_sort(...)` in a few places -- note that quite a few sort-algorithms make use of non-determinism to on average reach a better run-time.

That said, I'd still expect the output to be _correct_ w.r.t. the algorithms used, and _should_ only differ in things that 'don't matter' (like both things mentioned before, rounding off and sorting -- among others).

(It still can be tested because of that, if you don't specify exact output, but parameters within which the output should lie.)

I'd _prefer_ a deterministic engine though. I'll bring it up with the others and I'll get back to you about what the result is of that discussion.

GregValiant, 2022-11-17:
_"...there is absolutely zero difference (to 8 decimal places) between the prints."_
As long as that statement remains true on a simple model like a cube I will be a happy camper.
If some curves deviate by a hundredth of a millimeter, I don't think I'll notice.
If one slice shows my cylinder as a many faceted polygon, and the next it comes out as a cube then we're going to have to sit down and have a serious talk.

@Jrribas I'll go ahead and remove the bug label as this is (currently) expected behavior.
@fieldOfView and @rburema are you good with that as well?

rburema, 2022-11-17:
@GregValiant Yes, agreed. I'll close this now then.

(Am amendable to reopen if anyone else in the discussion has a different opinion, but I don't suspect that's the case right now.)

fieldOfView, 2022-11-17:
> Am amendable to reopen if anyone else in the discussion has a different opinion

Not me. The original issue was "not a bug" (the Search and Replace postprocessor has nothing to do with the changing output), and there's an internal ticket about determistic-ish-ity.

GregValiant, 2022-11-17:
_"deterministic-ish-ity"_
Ouch.  I was struggling with "Voronoi decomposition" ya know.