Ultimaker/Cura #13838: Layer hight - slicer settings issue

deffus, 2022-11-16:
### Application Version

5.2.0

### Platform

OSX

### Printer

Reality CR-10

### Reproduction steps

setting layer hight to "dynamic quality"
Adjusting material temp (215 on this case) 
Adjusting bed temp (55) 
hit slice

### Actual results

layers are sliced at 0.2 rather than 0.16 that the profile suggest they will.

### Expected results

0.16 layer hight and accurate time estimates 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10025221/cura.log)
[CCR10_Ratchet_Fidget.3mf.zip](https://github.com/Ultimaker/Cura/files/10025225/CCR10_Ratchet_Fidget.3mf.zip)
[image]

deffus, 2022-11-16:
[image]

fieldOfView, 2022-11-16:
The "dynamic quality" profile enables the "Use adaptive layers" setting. With that setting enabled, Cura varies the layer height around the layer height setting.

See https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/experimental/adaptive_layer_height_enabled.md

GregValiant, 2022-11-19:
@deffus can I close this?
As @fieldOfView explained the Adaptive Layer setting is enabled.  It appears to be a coincidence that your layer height was 0.16 and the "Adaptive Layers Maximum Variation" was 0.04 which adds up to 0.20 (your initial layer height).  Since in the case of this model there was no reason to "adapt" the layer height it went to .20 and stayed there throughout the gcode.

deffus, 2022-11-19:
Thanks both , I learned a thing and much appreciated :)

closed 2022-11-19 by deffus