Ultimaker/Cura #13839: Slicing Failed

DragonDon, 2022-11-16:
### Application Version

5.1.0

### Platform

MacOS 10.15.7, Nvidia GeForce GT 650M 1GB (mid-2012 MacBookPro)

### Printer

Tronxy XY-2 Pro

### Reproduction steps

I loaded up the Pectoral Plate from this file (https://www.thingiverse.com/thing:2667498), placed it flat on the bed, choose Normal Quality, 2.0, tree supports, no plate adhesion.

### Actual results

a 'slicing error' showed up with the recommendation to report a bg.

### Expected results

the STL should have been sliced with tree supports.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]
[cura.log](https://github.com/Ultimaker/Cura/files/10025294/cura.log)
[TXY2PROPRO2EXY3SE_pectoral.3mf.zip](https://github.com/Ultimaker/Cura/files/10025302/TXY2PROPRO2EXY3SE_pectoral.3mf.zip)

GregValiant, 2022-11-16:
Thanks for the report.
The slice failed for me as well.  Rotating the model 90° on the build plate allowed it to slice.  There are several reports about that particular bug.
That model is watertight but has some minor errors.  MS 3D Builder was able to repair it.
Below is your un-repaired model rotated 90°.
This appears to be a duplicate of a few other reports.  The Cura team is aware of this particular bug.
Let us know if that workaround helped get it to slice.
[image]

DragonDon, 2022-11-16:
Thanks Greg.  I ended up flipping the model so that the ‘top is down’ and it sliced that was as well.  Glad to be of help.

closed 2022-11-16 by DragonDon