Ultimaker/Cura #13841: Warning on slice if model z != 0

DavidCC00, 2022-11-17:
### Is your feature request related to a problem?

I'm frustrated when I forgot to zero a model so it's below he build plate, and I print a perfect model missing except it's missing the first 3% of the model.

### Describe the solution you'd like

When you press slice, if the model z != 0 (it's below the buildplate, or floating), it gives you a message.  

### Describe alternatives you've considered

As an alternative, we could work on cybernetic implants for me, so I never forget important steps to processes.  

### Affected users and/or printers

I think it affects all humans and most lizardmen secretly controlling the government.   It affects printers with a 3rd axis.   Inhabitants of Flatworld unlikely to notice. 

### Additional information & file uploads

Someone swapped my brain with a potato.

GregValiant, 2022-11-17:
Under "Preferences / Configure Cura" is a setting for "Automatically drop models to the build plate".  When it is turned on it will remember for you.  Of course you would have to remember to turn it off when you don't want it.

Please file another feature request if you want the "Turn it Off" reminder added.  The _Turn Things On_ group refuses to work on _Turn Things Off_ requests.  They're zealots.
I was going to request a "Hey dummy - move the Z seam" feature but it keeps slipping my mind.

Is Flatworld the same as Discworld?  I'm a fan of Sir Terry Pratchett.

MariMakes, 2022-12-06:
Hey @DavidCC00,

👋 I've never met someone with a potato as a brain. 

@GregValiant is right! We have a setting, Automatically Drop Models to Buildplate to help with this problem for models that are floating above the buildplate.
[image]

You are correct that it does not work if a model is dropped into the build plate. But there is also this different color in the prepare stage that will show you which surface is under, on, or above the build plate. 
[image]

I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞

MariMakes, 2022-12-09:
Hey @DavidCC00,

I brought your issue up to the team.
Since there are other indicators, to prevent user error, we believe we will not work on this anytime soon and will therefore defer this issue.
Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

I hope you understand.
Thanks again for your report.

DavidCC00, 2022-12-13:
> Is Flatworld the same as Discworld? I'm a fan of Sir Terry Pratchett.

I am too.  However, in this case, I actually meant to say Flatland, which is a writing about a land of 2D creatures. 

> 👋 I've never met someone with a potato as a brain.

Spend a few days working I.T.  then.   Interacting with end users is probably why I think stuff like this matters.  TBH, so far, missing a couple layers from my models hasn't actually ever been a problem and now that I know it's an issue, yes, it's easy to keep an eye on in the future. 

Yes, I'm aware of the setting.  The scenario I ran into is actually printing a large flat object (A gridfinity baseplate if you know what that is).  The software incorrectly decides to drop it below the build plate.  When you select "center" it'll correctly put the Z to 0 for enough frames for you to see it flash, and then it drops it below the Z again.  Once I manually set the Z to 0, "center model" stops trying to force it below the build plate.  I'm guessing the problem has to do with the algorithm to estimate best contact and this model (and one other I've seen) are edge cases.  

Since a lot of models don't seem to have very flat bottoms, I guess I had assumed that the light-red shading meant that part was touching the build plate, and not that it was beneath it.  

But you're right, I'm probably just far too stupid to use the software.  I'm unsure why you would suggest someone as stupid as me to contribute in any way.

closed 2023-12-28 by github-actions[bot]