Ultimaker/Cura #13842: Resize a model on one axis, *and only one direction*

altimac, 2022-11-17:
### Is your feature request related to a problem?

My only frustration with Cura is the resizing way it works:
 - I'm UNselecting "Uniform" to resize on only one axis
 - I drag the desired axis "cube" at the extremity of the axis to resize.... and the model is resized on BOTH sides of the axis !!!!
 => I would like the resize to only be applied in the direction I drag to.
This is particularly irritating when placing (moving) an Eraser exactly where I want to, but then need a minor adjustement of the size in a particular face of the Eraser cube. I resize it and boom, the Eraser is then adjusted both sides, making the precise initial adjustment pointless.

### Describe the solution you'd like

I would agree to press a specific key (option, shift, command or whatever) to modify the current behavior. I'm using a Mac so please be careful about conflict with other Mac standard behaviors.

**Please have a look at the attached video explaining the problem and the possible solution**

### Describe alternatives you've considered

Reajusting the initial placement over and over again

### Affected users and/or printers

I think every user will benefit from this yes.

altimac, 2022-11-17:
attached video too large. Will upload a smaller one today.

altimac, 2022-11-17:
video attached

MariMakes, 2022-12-09:
Hey @altimac,

I brought your issue up to the team. 
They really enjoyed your report and video. 

Cura is not CAD software but a slicer. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.
I hope you understand. Thanks again for your report.

closed 2023-12-23 by github-actions[bot]