Ultimaker/Cura #13843: the screen stay on slicing

jclien2703, 2022-11-17:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Alfawise U20 One

### Reproduction steps

hello, 
I have a problem with cura 5.2.1, I can't slice.

I load my files and i click on slice button

### Actual results

The screen stays on slicing.
[image]

### Expected results

I have to close cura because even the cancel button does not work

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I also have version 4.13.1 and I have no problem with this version.
[image]

I tried to do a complete reset of the cura 5 installation, but I have the same problem.

The problem comes from cura 5.
what can I do ?

[cura.log](https://github.com/Ultimaker/Cura/files/10029280/cura.log)

GregValiant, 2022-11-19:
Thanks for the report.
There has been a bug reported that can be fooled by rotating and/or moving the model on the build plate.  See if doing that frees up the slicer and let us know.
It my be a model problem though so load a model that shows the fault and use the "File | Save Project" command so create a 3mf file.  You will need to zip the 3mf file and post the zip folder here.

SharlSherif, 2022-11-22:
I have the same exact issue with v5.0.0 and the latest (5.2). I tried removing it entirely, restarting and then installing it again but that did not fix it. I also had a few friends have the same issue as well. The fix is using v4.13.1. I also have a custom printer profile if that makes a difference.

closed 2022-12-04 by github-actions[bot]