Ultimaker/Cura #13844: Crashes every time Mac sleeps at night

iamSteveRoberts, 2022-11-17:
### Application Version

5.2.1

### Platform

MAC OS

### Printer

Ender 3 V2 Sprite

### Reproduction steps

Every night when I leave my MAC Mini M1 go to sleep the next morning Cura and closed unexpectedly.

### Actual results

Restart as normal

### Expected results

Normal Running

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10029927/cura.log)

MariMakes, 2022-11-30:
Hey @iamSteveRoberts,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

What an interesting report 🤩 
I believe we've had internal reports about this aswell.

I'm not a mac user and I shut down my laptop at the end of the workday. 
Can you help me understand why it's important to leave your Cura open overnight?

In your logs I can see:
2022-11-16 21:29:04,216 - DEBUG - [MainThread] cura.UltimakerCloud.UltimakerCloudScope.requestHook [30]: User is not logged in for Cloud API request to https://api.ultimaker.com/connect/v1/clusters?status=active
2022-11-16 21:29:04,231 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestError [369]: request[c171240d][get][PyQt6.QtCore.QUrl('https://account.ultimaker.com/users/permissions')][timeout=10][no-data] got an QNetworkReplyError Host account.ultimaker.com not found. The server returned: b''
2022-11-16 21:29:04,233 - ERROR - [MainThread] cura.API.Account.error_callback [373]: Request for user permissions list failed. Network error: NetworkError.HostNotFoundError

How do you start a printjob on your Ender 3 V2 Sprite?
Do you load them on an SD card or are you connected to your printer with for example OctoPrint?

lantren, 2022-12-12:
I have also had Cura 5.2.1 crash when waking my MAC Mini M1 from sleep if it was active when the machine went to sleep.
No program should crash for this reason.  "why it's important to leave your Cura open overnight?" is meaningless.

closed 2022-12-14 by github-actions[bot]