Ultimaker/Cura #13853: Part Cooling - Fan Voltage Control

pcharalambakis, 2022-11-19:
### Is your feature request related to a problem?

Is this available in a configuration somewhere after a CURA build has  been complied? If not i would like to ask for a config feature request if possible?

Problem description - 
when in CURA UI selecting 0%-100% for material part cooling its  mapping to a 0-255 GCode command and in my case;
my motherboard runs Repetier-Firmware (Chinese industrial motherboard) and is a GCode command with a range of - M106 S0-258

problem i'm seeing is 0-100 percent in the UI isn't giving my part cooling fan more than 22v dc and only when it is at 258 does it get the 100percent. 

### Describe the solution you'd like

 (could we have access to the mapping of 0-255 threshold for the percentage 0-100% offered in the UI?

### Describe alternatives you've considered

editing GCODE after its been sliced is annoying and post script options aren't profile based.

### Affected users and/or printers

a background configuration will allow affected users to alter the percentage mappings being sliced and outputted to GCODE.

GregValiant, 2022-11-19:
Have you tried enabling the Search and Replace post processor?  (Extensions / Post Processing / Modify Gcode / Add a Script - Search and Replace).
Search  = M106 S255
Replace = M106 S258
Use Regular Expressions = unchecked
That wouldn't change the lower percentages but you would get your 258.

This is from the RepRap Wiki for M106:
Snnn Fan speed (0 to 255; RepRapFirmware also accepts 0.0 to 1.0))
From the Repetier site for M106:  "Fan on speed = 0...255"
So your 258 is non-standard.  The voltage you measure at PWM of 255 is the average voltage and may reflect your meter not responding quickly enough to the modulation or there may be a voltage drop across some of the components.
Cura has a new setting in "Printer Settings" (available in the MarketPlace) called "Scale Fan Speed 0 to 1.  You might also try that.

pcharalambakis, 2022-11-20:
Hi Greg,

Thankyou for the information above the search and replace function did what i needed nicely. Its an interesting motherboard on a mingda industrial printer.  manufactured in 2016. i will in future replace the motherboard and overcome the non standard implentations of repetier. but in the meantime many thanks to you for your prompt reply you solved my problem.

closed 2022-11-20 by pcharalambakis