Ultimaker/Cura #13854: Software Locks up and turns to a White Screen

cht685, 2022-11-19:
### Application Version

5.0.0 and up

### Platform

Windows 11 Home/Version 10.0.22000 Build 22000

### Printer

Ender 3 Pro

### Reproduction steps

1. Opened Cura 5.2.1
2. This latest occurrence was that I clicked on the Infill tab. There isn't anything that I can do that will cause it to reoccur every time, it is random at best. Last time it happened because I tried to change the nozzle temperature.

### Actual results

Cura locked up and became unresponsive. Only thing that is "clickable" is the Minimize button. Once I minimize Cura, I can reopen it. When it Reopens, it is a total white screen of death. Nothing works, except to close it with the 'X" in the upper right corner.  I can reopen Cura and it will either run without issue for hours/days, or it will lock up again.

### Expected results

I expected the software to run and not lock up and become unresponsive because I clicked on a tab or tried to change a value in one of the boxes. This only became an issue with Cura 5 and later. Cura 14.3 was the last version that I was able to run without incident.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I realize that Windows 11 is not "fully supported" by Cura 5.x, but Cura 4.x runs just fine on my machines. Yes I have tried multiple computers, and no there is no difference. The software is broken when it comes to Windows 11. I really do not want to go to Prusa Slicer for my Ender 3's. Please help.

j-snijder, 2022-11-21:
same here since 5.1.0 and perhaps earlier but can't say because I have stepped from 4.8 to 5.1.0 and did not update the between laying versions

MariMakes, 2022-12-01:
Hey @cht685,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13820
We've been seeing more reports on Cura slowing down on Windows 11 devices. 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13820

closed 2022-12-01 by MariMakes