Ultimaker/Cura #13863: Infill Overlap Percentag Broken

dgrat, 2022-11-20:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

My Own

### Reproduction steps

Not needed

### Actual results

Overlap Percentage is -30%
[image]

The Overlap starts to barely detach from walls just over -40%
[image]

Needless to say that the standard +10% lead to massive artifacts 

### Expected results

Overlap should be zero at 0% and not ~-40%

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Untitled.zip](https://github.com/Ultimaker/Cura/files/10050261/Untitled.zip)

GregValiant, 2022-11-20:
Thanks for the report and the file.
You have the "Outer Wall Inset" at 0.10mm and that is pushing the walls back into the infill by that amount.

The line width of 0.40 comes into play as well.  Cura calculates to the end of the infill line as it is the centerline of the nozzle.  Beyond that you have the radius of the extrusion (1/2 your line width) and so although the extrusion "path" is set back "-30%" the plastic occupies a radius of 0.20.  The path ends at .12 from the edge of the inner wall extrusion but at the .2 radius to that and yes, there is still an overlap.  The way the numbers work out here, you would need an overlap setting of -50% to completely avoid any actual overlap of the plastic.  So the math in Cura is accounting for where the Infill path ends in relation to the Inner Wall extrusion.  It is also dependent on your "Infill Line Angle.

_"...+10% lead to massive artifacts"_  When you do the infill first then it hardens and yes, the nozzle is going to bounce off the ends.  I've always considered 3 walls as a minimum to avoid things like that.  I consider 4 layers as a minimum for the top and bottom layers.  Less than 4 and you start to pick up artifacts on the skins, or dependent on the color, you can see the infill through the skins.

In regards to the bug report, someone from the Cura team will take a look.  The radius of the end of the infill extrusions along with the infill angle would need to be taken into account.  For now, if you have the infill coming up to the walls at 45° as in the block then you would need to have -50% overlap to keep the plastic of the infill from the plastic of those inner walls.

dgrat, 2022-11-21:
I was not aware that that the inset is related with this issue when I made the screenshots. I usually always use this inset because it avoids tolerance issues in 99% of cases. It would be great if the infill overlap percentage would take the inset into account. I am aware of the fact, that Cura has dozens of settings which may influence each other, which makes feature changes probably quiet hard.

GregValiant, 2022-11-21:
This is in the gray area between a bug and a feature request.  It looks like the way the overlap math is working that it's OK so long as we users understand it.  Adding the "1/2 LW" radius at the angle of the particular infill line may be beyond what the developers are willing to consider.  In a simple pattern like Grid the intersection with the wall occurs at two distinct angles 90° apart.  Considering that with Lines a user could enter the line angle at [45, 60, 90, 125] (and make no mistake - someone will) it would seem to get unmanageable.  The slicing is already a lot slower than on previous versions.

dgrat, 2022-11-21:
@GregValiant Don't destroy my dream of a fix :D

GregValiant, 2022-11-21:
I'm not destroying it.  I figure your chances are just about even with your overlap percentage.

dgrat, 2022-11-25:
I tested a bit, even if the inset is zero, the overlap is there

GregValiant, 2022-11-25:
Yep.  Overlap needs to be -50% in order to have 0.0mm of overlap.