Ultimaker/Cura #13865: Cura unable to slice when loading a *.gcode file

dgrat, 2022-11-20:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Cura stops working if you load a gcode file (any). 
Even is you clear the models and load *.stl files into the scene. 

### Actual results

Hangs/Slices indefinitely

### Expected results

Should work again at least after deleting the *.gcode file

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10050294/cura.log)

GregValiant, 2022-11-20:
Thanks for the report.
I have had this happen but I can't "make" it happen on purpose.  I had this problem occur this morning but after closing Cura (necessary since it wouldn't slice) and restarting it, it has been fine.
Someone from the Cura team will take a look.

MariMakes, 2022-12-02:
Hey @dg,

Thanks for your report 👍

In your logs I can see: `GCodeReader.FlavorParser.processGCodeStream [398]: Encountered a unknown type (CUSTOM VALUE) while parsing g-code.`

Could it be that you are using special settings, a different fork of cura, or printer definitions in the gcode that you have loaded?
Would you be able the gcode you wanted to load?

Thanks 🙏

GregValiant, 2022-12-15:
That is correct.  The Gcode reader doesn't display G2/G3 lines but I haven't had a problem loading an Arc Welded gcode file they just don't display as they should.
I have found the same to be true when loading gcodes sliced with other slicing software.  Cura's gcode reader expects to see certain lines in the gcode and they aren't there.  Like with Arc Welded file, they load but they don't display correctly.

dgrat, 2022-12-15:
Loading this, clearing plate and loading a stl works first time in 5.2.1. 
Doing this two times breaks slicing. 

[arcs_2.zip](https://github.com/Ultimaker/Cura/files/10240440/arcs_2.zip)

GregValiant, 2022-12-16:
I can't duplicate this.  I must have opened 20 STL's and gcodes today in various orders and I haven't had a problem.
MariMakes is in Europe so she probably won't get to this until at least Monday.

dgrat, 2022-12-16:
[Captures.zip](https://github.com/Ultimaker/Cura/files/10249069/Captures.zip)