Ultimaker/Cura #13866: Bug

Desobi31, 2022-11-20:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Longer LK 5 PRO marlin 2 (custom from github)

### Reproduction steps

Открываю 80 процентов моделей, отправляю на печать и они снизу справа идут рывками(точками): обычно модели маленьких кругов(3 мм) модели абсолютно любого содержания и только ближние 2(в моем последнем коде их 4). С левой стороны на моделях такие же отверстия и печатаются сразу линиями, а справой точками

### Actual results

Код не любит один и тот же угол(ближний правый ко мне) печатать на любых моделях: координаты стола начинаются снизу слева - печатает маленькими точками снизу справа вместо линий , дальше идет печать как положено движениями. Слева все в порядке. На видео видно как портится качество печати и падает скорость

https://user-images.githubusercontent.com/95028894/202922789-ea74b368-38e2-433a-af47-df765f9caf3a.mp4

### Expected results

Начал печать и подождал 2 минуты

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[1.zip](https://github.com/Ultimaker/Cura/files/10050561/1.zip)

GregValiant, 2022-11-20:
Thanks for the report.
My printer never does that and so I know I won't be able to reproduce it.  Are you printing via the SD card or are you using Octoprint or the USB?

I see in your project file that within the Mesh Fixes settings you have "Maximum Resolution" set to 0.25.  You might try setting that to 0.50 and see if it helps.  At .25 Cura is generating line segments a that are only .001mm long.  That's a lot of information for the machine to process.  Why it's different left to right I don't know.  AT 0.50 Maximum Resolution the shortest line segment is .02.

Someone from the Cura team will take a look.  I'm sure the gcode from left to right is correct as no one else has had this problem so it's likely your printer is either having issues with the gcode, or something is going on when it prints on that side of the bed.
If you move all the models over to the left does it still behave that way?

Desobi31, 2022-11-21:
> Thanks for the report. My printer never does that and so I know I won't be able to reproduce it. Are you printing via the SD card or are you using Octoprint or the USB?
I'm printing from an SD card. If you put a simple object, basically this does not happen. The line width of 0.25 is chosen due to the fact that the real nozzle on the printer is set to 0.5. Basically, this begins to happen on more complex models than, for example, a simple cube (with circles)
> I see in your project file that within the Mesh Fixes settings you have "Maximum Resolution" set to 0.25. You might try setting that to 0.50 and see if it helps. At .25 Cura is generating line [...]
 Thank, I'll try