Ultimaker/Cura #13867: problem with windows 11

Luca677099, 2022-11-20:
### Application Version

5.2.0

### Platform

Window 11

### Printer

Anet et4 pro

### Reproduction steps

The program is very slow especially when I move an item or in the printer management. 

### Actual results

nothing

### Expected results

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

j-snijder, 2022-11-21:
Same here, since v 5.1.0, somehow the devs added too much functionality ? or complexity ? and now cura is worse then ever.
It was the best slicer ever, but now I'm sorry to say its no longer, and in my case scenario not even usable anymore.
Slow as a snail on the southpole and freezing from time to time so I have to use programm manager to shut it down.

For the time beeing I'll switch to crealty slicer v 4.8.x which does not have these problems and in fact is an older cura instance.

Please direct your attention to this and make it top priority, as lots of us users are using this fantastic slicer.

MariMakes, 2022-12-01:
Hey @Luca677099,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Thanks for your report 👍

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13820
We are investigating what is going on.

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13820

closed 2022-12-01 by MariMakes