Ultimaker/Cura #13868: Reload models - Doesn't reload

tlhintoq, 2022-11-20:
### Application Version

5.2.1

### Platform

Windows10x64, 21H2

### Printer

Ender3pro

### Reproduction steps

1. Build a layout.  Add models.
2. Save as Cura project.
3. Update the original model.  Same path, same name, just updated in place.
4. Re-open the project.
5. Select "Reload all models"
6. Nothing updates

In this example the early version had text embossed on the top face.  The updated version removed the text.  After "Reload all models", we can see the old version still on the layout.

[image]

### Actual results

In this example the early version had text embossed on the top face.  The updated version removed the text.  After "Reload all models", we can see the old version still on the layout.

### Expected results

The models should have updated to the new version of the STL

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

YOu want the project file... but you don't support the project file type on upload.  Really?
[2xDualOutletBoxBack.zip](https://github.com/Ultimaker/Cura/files/10050850/2xDualOutletBoxBack.zip)

MariMakes, 2022-12-02:
Hey @tlhintoq,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

It took me a couple of tries, but it seems that if you are in the same session and you load the models from an .stl you are able to reload the models with a pop-up or with the reload options in Cura 5.2.1
[image]

But if you load the models from a .3mf project file it seems to have lost the link to the original .stl path.
I'll have to check with the team if this was intentional or not since .3mf project files are a common way to share models so I expecct that the link is broken after export and starts living inside of the .3mf. If that makes sense 🤔 

Anyways, I'll come back with an answer. 👍

MariMakes, 2022-12-09:
Hey @tlhintoq,

I brought your issue up to the team. 
Models saved in a .3mf lose their connection to the original stl.
If you want to keep working with this 3mf file, I suggest to load the file, removing the model from the buildplate, and add it the new model again. 

I'll be closing the issue since it's not a bug in Cura.
I hope you understand.

closed 2022-12-09 by MariMakes

terachip, 2023-08-02:
@MariMakes @tlhintoq 

Adding a comment because this is definitely a bug, or at least the next step wasn't considered.  If you have a complex part and add support blockers (or to provide per model slicing settings) it stands to reason that you'd want to save it as a project in case you want/need to change settings later.  But if you need to change the part itself, without a link to the original part, you have to do everything over with your support blockers or per model settings.

So without being able to either, 1) reload the STL like Cura normal does (if you keep the STL link), or 2) being able to manually point Cura at a different file (sure, leave the link broken, but why not the option to re-link it), you're unable to revise the original geometry.  It's not even possible to delete the STL to add a new part and keep the support blockers; the support blockers get deleted as well.

Or is there a plugin that provides this option?

tlhintoq, 2023-09-30:
@terachip @MariMakes 

Terachip you are exactly right... if you have a part/layout with actual needs like blockers, custom supports, rotations and so on, then you are hosed.  Deleting the part and re-importing is exactly what the user is trying to avoid by wanting to use "re-import" or "reload model".

Basically, the more you *need* to be able to use re-import the less likely you are *able* to do it.
If you have a totally brain-dead layout of just drop the part and print then you don't really need to re-import do you?  Deleting and drag-n-drop is fine.

So basically - this feature is only a one-session-use ability.  As long as you have that specific Cura session open when you make a change to the part then you're in luck.  I guess you have to dedicate a laptop to just running that one session for that one model, forever. That way when you want to make a change, like update the model for personalization then reprint you can do that.  Might as well just run the printer from a dedicated laptop over USB, I guess.

{sorry if my frustration with this being called 'not a bug' is showing.}

fixundfertig123, 2023-10-01:
Hi,
absolutely agree with @tlhintoq and @terachip. It may no be a bug, because it is by design. But maybe your design decision has not considered the above use case. IMHO, for any serious application use this is a must. Project's are great in general. But development and evolution in specs means most likely changes. And your current project approach only works for once and forever finished designs. I have hundreds of smaller or larger designs, but hardly any is finished forever. E.g. Inventor or Solidworks offer the option to break/disable the link between a derived model or maintain it. Break it -> if you do want to freeze a design. Keep the link -> work on a not finished design.

So please @MariMakes , reconsider your decision. Thank you.

tlhintoq, 2023-10-04:
@terachip @fixundfertig123 

I'm going to give PrusaSlicer another go.  Its seen 3 updates since I last looked at it.  I know its supposed to have the same improved tree support logic that Cura does. And way too many FB posts about how it is solving lots of the niggly issues like idle oozing and stringing that no amount of playing in Cura seems to resolve.  Maybe that's just user noobie-ness.  maybe its not.  But going to give it another look since those developers seem to be taking user feedback into account.

jonathanmonette, 2023-10-05:
I totally agree, There should be a way to rebuild this link, or do a replace part and keep all the blocker or modifier that was previously added. very annoying.

fieldOfView, 2023-10-05:
The Mesh Tools plugin will let you replace a model on the buildplate with a model from disk.

fixundfertig123, 2023-10-05:
Hi @fieldOfView. Great tip and great module you developed! Looks very well thought through.

tlhintoq, 2023-10-05:
> The Mesh Tools plugin will let you replace a model on the buildplate with a model from disk.

That's freaking GOLD man.  You are a legend!
[image]