Ultimaker/Cura #13869: Ultimaker Cura 5.2.1 for Linux

vrtualjim1965, 2022-11-20:
### Application Version

5.2.1

### Platform

Linux

### Printer

Ender 3 pro

### Reproduction steps

1. Open .sti file
2. selected the Slice button
3. save to disk
4. selected the usb flash drive to copy to
5. name the file 
6. save
7. the file saves to the flash drive
8. removed flask drive from linux computer
9. insert into Ender 3 pro
10. select print
11. select udisk

### Actual results

no file found on the udisk

### Expected results

1. found print file
2. select print file and start print
[Fuse2.stl.zip](https://github.com/Ultimaker/Cura/files/10050891/Fuse2.stl.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

works perfect in windows version

MariMakes, 2022-12-08:
Hey @vrtualjim1965,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/11991
We have a ticket on our backlog with the intent to improve this behavior.
If you read through the ticket this might also be related to https://github.com/Ultimaker/Cura/issues/12659

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/11991
Thanks again for the report, it helps to have more examples when something goes wrong.

closed 2022-12-08 by MariMakes

MariMakes, 2023-03-31:
Quick update on our side. 

We might have a fix for this issue. 🎉 
It's a bit hard for us to test, but we have a nightly build available. 
Can you please try it out and let us know if it resolves your issue? 

[You can find the download links and the discussion here. ](https://github.com/Ultimaker/Cura/discussions/15098)