Ultimaker/Cura #13870: Option to add a small chamfer to the first few layers

lagnat, 2022-11-20:
### Is your feature request related to a problem?

First layer tends to have some "squish".  It's difficult to completely tune that out and sometimes it's desirable in order to maximize bed adhesion.  That usually means some sanding and cleanup is necessary.  While not a huge problem, it's time consuming and cab be difficult if the print has curves or tight turns. 

### Describe the solution you'd like

I'd like to be able to enable a print feature that adds a small chamfer to the first layers.  With a little tuning, one could get first layers to have sufficient squish and not end up protruding beyond the perimeter of the print.

### Describe alternatives you've considered

I am not aware of alternatives

### Affected users and/or printers

This would affect most people that have strict fit requirements or want higher quality finished prints. People who print high volumes would really benefit.

Asterchades, 2022-11-21:
Does the Initial Layer Horizontal Expansion setting. used with a negative value, not already do this? It only applies for a single layer, but if your mechanical Z offset is so far out that it requires multiple layers of "fade" you would be better served by addressing that rather than trying to compensate at the slicing phase.

lagnat, 2022-11-21:
I didn't think to try it and you're probably right on.  Closing.

closed 2022-11-21 by lagnat