Ultimaker/Cura #13871: Does not allow window resize  custom color picking in Materials.

roomonthethird, 2022-11-21:
### Application Version

5.2.1

### Platform

macOS Monterey 12.6.2

### Printer

Creality Ender 3

### Reproduction steps

1. Opened Preferences > Configure Cura > Materials
2. When adding custom material or editing a current one, clicked on the color.
3. Opens up color picker window that doesn't allow me to resize.
4. Unsure if custom color picker is still available as I cannot see or resize the window.

### Actual results

Cannot resize window.

### Expected results

Allows me to resize window to see full content of the subscreen.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[img](https://imgur.com/a/GJvVIDF)

MariMakes, 2022-12-08:
Hey @roomonthethird,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I've seen this issue myself as well! I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞

MariMakes, 2022-12-12:
Hey @roomonthethird,

I brought it up to the team and it should be resolved with this commit: https://github.com/Ultimaker/Cura/commit/7e8eca3b03516bc1b449a6064aaa4b711596582c

That means that you can expect it in the next Cura release.
I'll be closing this issue, thanks again for the report 👍

closed 2022-12-12 by MariMakes