Ultimaker/Cura #13875: Kobra MAX bug

Pryador, 2022-11-22:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Anycubic Kobra MAX

### Reproduction steps

Tried to slice my custom stl file.   I tried to flip it so that there was no support on the inside of the dome. This caused the error.  

### Actual results

Told me I had a bug

### Expected results

Slice 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Goose Vent.zip](https://github.com/Ultimaker/Cura/files/10063149/Goose.Vent.zip)

GregValiant, 2022-11-22:
Thanks for the report and the file.
That's a nice model and I didn't find any errors in it.  I don't see any way to print it without support though.

I rotated the model and it sliced OK until I had it upright like a bowl.  So I rotated it 90° on the build plate and it sliced.
This appears to be a duplicate of several other reports where good models don't want to slice when in a particular orientation.  The "rotate them some more" workaround seems to fool the bug a lot of the time.  This has nothing to do with your printer or printer definition file.

I'll mark this as a duplicate but I'll leave this report open as your model may be easier for the team to figure out since it is symmetrical about the Z.

Here it is sliced.  I'm pretty sure all those tall outside supports for the tops of the triangular pockets would fail.  

[image]

Pryador, 2022-11-22:
I am going to redo the model so I can print it as a dome with a cone on top
so I don't need the supports inside.  I appreciate the help and will let
you know if I have any more issues.
Thanks
John