Ultimaker/Cura #13876: Prime Tower/Ooze Shield issue

madurani, 2022-11-22:
### Application Version

5.2.1

### Platform

linux

### Printer

homemade 4-IN-1-OUT

### Reproduction steps

 Problem appear when is activate options "Dual Extrusion/Enable Prime Tower". Cura still use for building the first layer and some ring around in next layers of tower a extruder1(T0), although isn't use in printing of model.

For example: I print model, where are use only extruder3 and 4 (T2 and T3), but tower's first layer is printed with extruder1(T0). It has negative color mix effect in next layers,  because is used T0 extruder, altrought printed model not use it, but Cura add T0 to every layer.
 Extruder T0 is added too in Ooze Shield.

Some screen from Cura:
T0 black
T1 red
T2 white
T4 blue
[image]
[image]

### Actual results

 It has negative color mix effect in next layers. Unused color is added inertia in next layer to model.

### Expected results

Unused color/extruder shouldn't be use neither in tower neither in "Ooze Shield".

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[CFFFP_color2.zip](https://github.com/Ultimaker/Cura/files/10065369/CFFFP_color2.zip)

GregValiant, 2022-11-22:
Thanks for the report.
Unfortunately a gcode file doesn't show how you have Cura set up.  I can see what happened but can't tell why it happened.  We need a project file ("File | Save Project").

Have you tried disabling your unused extruders?
_"...negative color mix effect in next layers..."_ I have found that going from any color to white takes a lot more purge.  Your T2 might require more purge when it comes in.

madurani, 2022-11-22:
Yes I tried disable two unused extruders, but problem persist. Extruder (T0) stay in use.
Temporary solution was modiffi output gcode-file from Cura and change there T0 to T2:
`sed -i 's/T0/T2/g' CFFFP_color2_sed1.gcode`

_We need a project file ("File | Save Project")._
added:
[CFFFP_color2 (1).zip](https://github.com/Ultimaker/Cura/files/10067019/CFFFP_color2.1.zip)

_I have found that going from any color to white takes a lot more purge. Your T2 might require more purge when it comes in._
Yes I know about it and it is setup in configuration via "minimal tower volume"

Please look on end of the first layer: nozzle finish with T0(black) on tower round , then do only one line on  "skirt" with T3(blue) and immediately begin on model with T3(blue) without cleaning. It isn't correct behavior. 

Finish black round:
[image]
Continue with one line of "skirt" with T3(blue) and  then immediately begin on model  with T3(blue) without cleaning 
[image]

GregValiant, 2022-11-22:
Thanks for the project file.
I fooled around with this and can't get Cura to ignore T0 unless I disable the Prime Tower.  If the Prime Tower is enabled then Cura doesn't appear to care if T0 is disabled or not, nor whether it is actually used or not.
The same problem occurred with a Geeetech A20T (3-in-1-out) and my own Ender configured as a 4-in-1-out.

I was able to get rid of it from the Gcode by doing sort of what you did and using Cura's Search and Replace post processor like this:
Search  = T0\n
Replace = T3\n
Use regular expressions = checked

Someone from the Cura team will take a look.

madurani, 2022-11-22:
I checked a few older version of cura, which a had on pc and a prime tower had same behavior on all. Then I checked old closed issues here, which were very similar with my problems. It look like, that it is old unsolved issue.

madurani, 2022-11-23:
I tried use other slicer tool under linux "PrusaSlicer". I point out, that i before never saw this tool. After some chaotic configuration I stayed surprised with a result. 
I printed same model on same printer (I notice, that printer is homemade and no exist any customized profile for this printer).
Model on picture above is from Cura slicer and below same from Prusa slicer. Quality from Prusa slider is more better and there isn't any problem to print with more extruders:

[image]

[image]

[image]