Ultimaker/Cura #13878: [Feat] command line arguments

hiaselhans, 2022-11-22:
### Is your feature request related to a problem?

I want to run cura from the command line to slice multiple stl's in a script.

### Describe the solution you'd like

In earlier versions of cura the command-line for CuraEngine was logged. That was a good way to get all the arguments needed. These days the log is empty, and it would really help to somehow export all the settings required to run curaengine.

### Describe alternatives you've considered

Possible solutions:

- write the command-line arguments to the log
- provide a way to "export" the current settings as e.g. json
- let CuraEngine pick the last used config as default

### Affected users and/or printers

Any user who wants to run batch jobs

MariMakes, 2022-12-06:
Hey @hiaselhans,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

This request is very much outside my field of expertise. 😖 
I'll bring it up to the team to see how they can help you out. 
Fingers crossed. 🤞

MariMakes, 2022-12-13:
Hey @hiaselhans,

I brought your issue up to the team and it seems to be an unintentional change, so I changed your feature request into a bug. 
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9986

Thanks for the report! 👍

MariMakes, 2023-02-06:
Hey @hiaselhans,

Quick update, we found what caused the issue and resolved it. 
You can find the change here: https://github.com/Ultimaker/CuraEngine/commit/530e25690428e6013cba1ac62646ed6624e5740b

It will be shipped with the next Cura Release.
Thanks again for reporting! 💪

closed 2023-02-06 by MariMakes

victornpb, 2023-02-23:
> In earlier versions of cura the command-line for CuraEngine was logged. That was a good way to get all the arguments needed. These days the log is empty, and it would really help to somehow export all the settings required to run curaengine.

I'm sorry to resurrect this issue, but I'm trying to do the same, can you share some instructions how to correctly find the parameters and pass it along to the cli?

MariMakes, 2023-02-27:
Hey @victornpb,

Did you try the new Cura 5.3 Beta? It contains the fix for this. 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0-beta.1

Can you help me understand why you are running CuraEngine from the command line? 
I would love to know more about your project, maybe we can help.

victornpb, 2023-02-27:
> Can you help me understand why you are running CuraEngine from the command line? 
> 
> I would love to know more about your project, maybe we can help.

I’m trying to do some automated slicing, because my workflow is usually like this

1. open an stl
2. slice it for ABS save to usb/abs/name.g
3. slice it for PLA save to usb/pla/name.g
4. slice it for PETG save to usb/pla/name.g

I work this way because I have a usb drive ready to print anything, i just load the filament and select the appropriate folder file and hit print. no computer involved

i would also like explore some other ideas.

but i haven't managed to slice a stl successfully with the cli yet

rburema, 2023-03-01:
@victornpb 

I should not that the official support status of the cli is something like 'when we get to it'. Aka; not properly officially supported.

That said, writing up proper documentation is a thing that should happen sometime.

In the meantime, there's quite some issues in the engine issues-list that provide examples. For a relatively recent one, [see here](https://github.com/Ultimaker/CuraEngine/issues/1660#issuecomment-1336511892).