Ultimaker/Cura #13879: City of Tarok Road Side Inn: tower_room_top.stl

salvatoretramuto, 2022-11-22:
### Application Version

5.2

### Platform

macOSMonterey 12.6

### Printer

ANYCUBIC S

### Reproduction steps

1. updated download file model
2. sliced file model

### Actual results

Slicing Failure: Slicing failed with an unexpected error. Report bug.

### Expected results

file to slice like remainder of model

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-11-23:
Thanks for the report.
Without a Project File ("File | Save Project") and a cura.log file (@ "Help | Show Configuration Folder") there can be no investigation.  Zip those files and post the zip folder here.

salvatoretramuto, 2022-11-23:
[tower_room_top.zip](https://github.com/Ultimaker/Cura/files/10075635/tower_room_top.zip)

salvatoretramuto, 2022-11-23:
Please find Zip of file. Thank you and Happy Thanksgiving!

GregValiant, 2022-11-23:
[image]

The model doesn't appear to have any errors and sliced for me.
This is why we ask for a "Project File".  With nothing to check but the model I find it to be OK and I can't duplicate the problem.  It may be your printer, or your settings, or a Cura bug.

closed 2022-12-14 by github-actions[bot]