Ultimaker/Cura #13881: Brim stops Slice

ChaosLord76, 2022-11-22:
### Application Version

5,2

### Platform

MACOS

### Printer

Ultimaker 2+ Extended

### Reproduction steps

i have a Modell which fits in the building Area although it ist only some mm smaller. The contact Areo with the headbed is in the mittel and only 5x5cm large. When I enter a Brim lager than 3 cm the slicer does not start because the model then does not fit in the Area. when brim is smaller it works. The interesting part is, that the support-Structure is far away from the 3cm brim so it is within the area 

### Actual results

nothing you can not slice

### Expected results

I guess that the area the brim takes is calculated from the largest positions of the model and not from the contact area with the heated. Normally I would think, that the slicer should use only the contact area to calculate the brim and depending on this set the slice key free
[image]
[image]
[UM2E_Geformter_Korper_21-11-22_21:32:42.3mf.zip](https://github.com/Ultimaker/Cura/files/10071726/UM2E_Geformter_Korper_21-11-22_21.32.42.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have added screenshots of the Model as I placed it bevor and after slicing and my project File (and sorry for my bad school English)

GregValiant, 2022-11-23:
Thanks for the report.
You are correct that Cura calculates the footprint of a model by the "shadow" of the model and not the footprint itself.  This has been brought up before and was explained by a developer.  See Cura #10209 and refer to Ghostkeeper's comments there.
In that instance the problem was "Deferred".

Someone from the Cura team will respond.

ChaosLord76, 2022-11-23:
hi,

the issu you refer to is, that die larger areal in the issue was under the bild-Table, that ist not the problem here. All areas I refer to are above the build Plate, but you might be right, because the Issue says that the Brim is calculated on all Layers... I guess that is exactly what I see here, but I thing that it is interesting to solve this because in this case the brim only need to be calculated on the first few layers so it might even be easier to do than to use the whole model, but I have never looked inside the code of course :)

Letzt See what the Cura Team will say to this 

Steve