Ultimaker/Cura #13882: (4.12.1) Support type not updating

mayaderen, 2022-11-23:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Creality Ender #3

### Reproduction steps

Changed the support type
Sliced file

### Actual results

The preview support type was not the support type selected

### Expected results

The support type changes in the preview

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Project bug.zip](https://github.com/Ultimaker/Cura/files/10072840/Project.bug.zip)

Asterchades, 2022-11-23:
Is this still the case in 5.2.1?

Cura 4.12.1 is nearly ten releases behind the latest official build. It's quite possible your issue has already been addressed by one of those updates, and you would need to install a newer version anyway to receive any fixes should it not be.

GregValiant, 2022-11-23:
@mayaderen I have 4.12.1 loaded and I can't duplicate the problem.  When I select "Tree" it shows tree supports and when I select "Normal" it shows normal supports.
If you don't want to move up to 5.2.1 as @Asterchades suggests then you might try 4.13.1.  It was very good.

mayaderen, 2022-11-23:
@Asterchades @GregValiant 
Thank you for the suggestion, I have tried using 5.2.1

I should have also specified that I can switch between normal and tree supports but when I am using normal supports, only grid type of supports will be used, even when I select zig zag or grid.

GregValiant, 2022-11-24:
Check your "Support Infill Line Directions".  If it is empty brackets "[ ]" then you should see lines and/or zigzags.  If it is something like "[45, 135]" then lines and zigzag will look just like Grid because the lines change angles every layer and in the case of 45, 135 they are 90° apart just like Grid.
( Thanks @Asterchades )

Asterchades, 2022-11-24:
I think I see what's happening here. Capture (1).png in your included project file shows two different types of support structure - the base supports themselves (teal), which appear to be configured as a 45° Zig Zag, and the Support Interface (pastel blue), which is set to Grid.

My guess is that you're adjusting the Support Pattern and expecting the wrong one to change. This setting only applies to the base support structure (teal). You need to adjust the Support Interface Pattern to change the other visible pattern (pastel blue). You will either need to adjust the setting visibility to show this setting or search for it by name. There are other Interface-related settings which are also hidden, including whether you want it enabled or not, how thick it should be, what density it should use, and so forth.

GregValiant, 2022-11-24:
Good call there @Asterchades .

closed 2022-12-14 by github-actions[bot]