Ultimaker/Cura #13885: Cura Redownloads constantly

CornyBirdy, 2022-11-24:
### Application Version

5.2.1

### Platform

windows 10

### Printer

ender 3 v2

### Reproduction steps

every time i try opening a 3d file and it keeps reinstalling even though it`s already downloaded

### Actual results

nothing happens but its just really annoying how it redownloads it self again 

### Expected results

it just opens up like normal

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nothing here...

fieldOfView, 2022-11-24:
You have associated your 3d files (stl, I assume) with the Cura installer instead of the Cura executable. This is not something the Cura installer itself does.

To fix this, open the Windows menu and search for "Choose a default app for each type of file". 
Scroll down to .stl (it is a LONG list), press the icon beside it and select "ultimaker-cura.exe".

MariMakes, 2022-12-01:
Hey @CornyBirdy,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Thanks for the suggestions @fieldOfView, I believe that should've resolved it as well.

Do you perhaps have a screenshot of what you are seeing?
How many versions of Cura do you have installed on your computer?

closed 2022-12-16 by github-actions[bot]