Ultimaker/Cura #13887: Stringing since 5.2.1

sonstiges, 2022-11-24:
### Application Version

5.2.1

### Platform

Windows 10 64 bit

### Printer

Ultimaker S5

### Reproduction steps

I have tried to do a stringing test several times with the same profile as version 5.0.0. No matter which settings I used, the result was full of stringing, even with retraction.
When I printed the same file with the same material and the same profile with version 5.0.0, there was no stringing.
I have found several unanswered posts on the internet in forums about this, describing exactly the same phenomenon. So I am not the only one with the problem.

### Actual results

Cura 5.0.0 = a little bit stringing
Cura 5.2.1 = massive stringing, no way to get rid of it.

### Expected results

no stringing at all.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no additional information

GregValiant, 2022-11-24:
Thanks for the report.
This is the first report I've seen here about this.  I've had no problems with my Ender but UM machines are different.
Use the "File | Save Project" command to create a 3mf file of the problem model and zip the file.  Post the Zip folder here.  Maybe you could add one of those bad gcode files to the zip as well.

j-snijder, 2022-11-24:
Now you mention it, I experienced lots of stringing and blobs too since v??? could be 5.1 or later.
I thought it was my filament thats wetter then normally because wether changed and we are in mor moisture days now.
Sorry that I can not send some project file of my earlier made prints.

GregValiant, 2022-11-25:
I review a lot of gcodes and project files and I haven't seen any issues with them or with my own projects.  Settings like "Retraction Minimum Travel" and "Max Comb distance with no Retract" come into play as well.

If you have a fancy text editor like Notepad++ one of the features is it will let you count how many occurrences of the search term are in the file.  Search for "00 E" (00spaceE) and it will find all the retracts and primes and let you know how many are in the file.  Divide by 2 to get the number of retractions.

sonstiges, 2022-11-26:
I would like to formally apologise. In the meantime, the error has also occurred in Cura 5.0.0 and Cura 4.13. 
I have no idea where it comes from. The problem is that the nozzle leaves a blob at the end of a line when it retracts. 
I have changed the needle, cleaned the feeder and taken the whole printer apart. I can't find the fault. It occurs with 5 different filaments.
Since I can't set retraction-speed faster than 45mm/s, I probably won't be able to fix the problem. I also have no idea where this is suddenly coming from.

I apologise again for taking up your time.

closed 2022-11-26 by sonstiges