Ultimaker/Cura #13889: slicing erro

doommaker79, 2022-11-24:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Anycubic i3 Mega S/Pro

### Reproduction steps

Set all the print settings, tried to slice, but it fails at about 60%

### Actual results

slicing error

### Expected results

successful slice

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.zip](https://github.com/Ultimaker/Cura/files/10086941/cura.zip)

doommaker79, 2022-11-25:
Under MESH FIXES -> Maximum Resolution: if it is set to 0.01mm, slicing crashes. If I set to default 0.5mm all is OK!

MariMakes, 2022-12-01:
Hey @doommaker79,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Thanks for your report 👍

I'm getting the same slicer error as you.
But... I also managed to slice it when I moved the model a little. 
Away from the center of the buildplate.
[image]

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13643
I'll have someone from the team take a look.

Thanks for the report!

closed 2023-01-27 by casperlamboo