Ultimaker/Cura #13890: Cura Slicer Crashes

pepino73, 2022-11-24:
### Application Version

5.2.1-mac

### Platform

macOS

### Printer

Snapmaker 350T

### Reproduction steps

Model eines fachen Reliefs geladen,
Druckenstellung o,o8 eingestellt
[image]
[image]

### Actual results

Absturz

### Expected results

Slicen

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[image]

MariMakes, 2022-12-01:
Hey @pepino73,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

This mesh seems faulty, but it's hard to judge with just a screenshot.

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools
If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/.

pepino73, 2022-12-02:
Danke, ich habe das Problem gelöst.

MariMakes, 2022-12-02:
Thanks for getting back to us. 
Relieved to hear it's resolved. 

I'll close the issue.

closed 2022-12-02 by MariMakes

pepino73, 2023-11-15:
Warum funktioniert das Zoomen in Cura nicht.
Ich habe einen Mac /13.6.1 (22G313)
Mit freundlichen Grüßen
J. Fiedler

> 
> Danke, ich habe das Problem gelöst. 
> 
>> 
>> 
>> 
>> Hey @pepino73 <https://github.com/pepino73>,
>> 
>> Welcome to the Ultimaker Cura Github 🚀
>> Sorry, it took us a while to get back to you 😞
>> 
>> This mesh seems faulty, but it's hard to judge with just a screenshot.
>> 
>> Do you have a project file for us? It contains the printer and settings we need for troubleshooting.
>> To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.
>> 
>> You'll need to open Github in your browser to attach any files to your ticket
>> You can click the view it on GitHub link at the bottom of your email to open GitHub in the browser.
>> 
>> Have you tried doing a quick mesh fix with the Meshtools Plugin?
>> You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools
>> If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/.

￼

        J.Fiedler
portfolio.adobe.com/c4979d3d-1e90-4896-a415-c82c75d4b809/editor/
portfolio.fofcommunity.com/nanos-portfolio