Ultimaker/Cura #13892: Slicer crashed

eavakyan, 2022-11-24:
### Application Version

5.1.0

### Platform

MacOS

### Printer

Ender 3 V2

### Reproduction steps

Tried to slice an stl
[Fuselage 4 B L ribs.stl.zip](https://github.com/Ultimaker/Cura/files/10087189/Fuselage.4.B.L.ribs.stl.zip)
[CE3PRO_Fuselage 4 B L ribs.3mf.zip](https://github.com/Ultimaker/Cura/files/10087194/CE3PRO_Fuselage.4.B.L.ribs.3mf.zip)

### Actual results

Slicer error, model not sliced.

### Expected results

Model properly sliced for printing

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Attaching the project file, log, and the stl of the model that was being sliced.

[cura.log.zip](https://github.com/Ultimaker/Cura/files/10087202/cura.log.zip)

GregValiant, 2022-11-26:
Thanks for the report.
I wasn't able to load the 3mf file in Cura so I didn't get your settings.  MS 3D Builder wouldn't open it either.

The STL wouldn't slice for me either using 5.2.1.  I rotated the model 90° CCW and it sliced.
This bug has been reported numerous times in 5.1 and although it was worked on, it is still present (somewhat) in 5.2.1.  The bug can often be fooled by rotating and/or moving the model on the Cura build plate.

I'll mark this as a duplicate "slicing error" and close this as the Cura team has a lot of models that showed this behavior.  The problem should be fixed in 5.3.
If you feel that closing this is premature let me know here and I'll re-open it.

closed 2022-11-26 by GregValiant

eavakyan, 2022-11-26:
I tried the same thing after sending in the report - rotated it 90 degrees
and was surprised that it sliced just fine. Thanks for checking and I look
forward to seeing 5.3 in action.