Ultimaker/Cura #13899: Machine Settings "Compatible Filament Diameter" doesn't update Printer Settings "Diameter" correctly

GregValiant, 2022-11-25:
### Application Version

5.2.1

### Platform

Win 10

### Printer

Any

### Reproduction steps

Go to Machine Settings and change the Compatible Filament Diameter to a non-standard number.  I used 5mm
Go to the Printer Settings and check the "Diameter" setting.  It will not be the same and Cura will use the number in "Diameter" rather than the number in Machine Settings.   In this example the Diameter is 2.85.  It appears to set it to a standard diameter regardless of what is entered in Machine Settings.
[image]

### Actual results

Cura used the number in the Printer Settings Diameter box to calculate the E values.  The Printer Settings "Diameter" box does not update to the diameter entered into the Machine Settings "Compatible Filament Diameter" box and vice versa.
Here I entered a custom filament diameter into Printer Settings and it did not update the "Compatible Diameter" box in the machine settings.  In this case Cura did use the 12.0 diameter to calculate the E values for the gcode.

Here I have entered "12" into Printer Settings and Machine Settings did not update.
[image]

### Expected results

When entering a diameter into the Machine Settings, the Printer Settings should update and the opposite should also be true.  The number should then be used in the calculations.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Project file attached.
When changing the "Machine Settings / Compatible Filament Diameter" to 12mm the "Printer Settings / Diameter" updated to 2.85mm and used 2.85 in the calculations.
The discussion on the UM Forum is [HERE](https://community.ultimaker.com/topic/42151-bug-in-cura-51-material-diameter/#comment-315473).

[Filament Diameter Bug Report.zip](https://github.com/Ultimaker/Cura/files/10093205/Filament.Diameter.Bug.Report.zip)

fieldOfView, 2022-11-25:
> Here I have entered "12" into Printer Settings and Machine Settings did not update.

That is correct, and by design; the value in the Print Settings may override what is set elsewhere. The same is true for example in the Materials pane of the preferences; whatever print temperature is set there may be overridden by the print temperature in the Print Settings.

GregValiant, 2022-11-25:
Thanks.  If it is going to over-ride shouldn't it update the other setting box?  "Printer Settings" isn't installed by default and I don't think the poster over on the Forum was aware of it.  Maybe it's just a tooltip thing, but it isn't very clear at all.

If I set the "Filament Diameter" I would think it safe to assume that I have set the Filament Diameter.  Entering 12 (or whatever) into the box in Machine Settings and having the hidden box in Printer Settings change to 2.85 is a lot less than intuitive.

It should also work the other way.  If I set the diameter in Printer Settings then the box in Machine Settings should update to reflect that.

MariMakes, 2022-12-02:
Hey, you! 😉 

Thanks for your report 👍
It's odd that we have the Machine Settings / Compatible Filament Diameter but we don't tell people that its expression is only limited.

I'll bring it up to the team 🤞

Joeydelarago, 2022-12-09:
Hey, I'm having a look at this issue but I'm slightly confused about what the issue is.

I see that the user in the forums is setting the **extruder** material diameter, this works as I would expect. You can only use materials with this diameter (Tested on 5.1 && latest).

The printer settings plugin is reading the **printer** material diameter and setting this to a quality profile. The reason they don't match is the two settings are being read from different places.

A simplified order of precedence for the diameter setting will be like this 

1.  Quality Profile (Settings set with Printer Settings plugin)
2.  Extruder (The setting set in the machine dialog, extruders tab)
3.  Printer  (The setting **read** by the printer settings plugin, not where it is saved though)

My question is, what is happening when using the software that gives the impression that the material diameter is not being set correctly without using the printer settings plugin?

GregValiant, 2022-12-09:
@Joeydelarago I can't speak for the poster, but I'll try to describe what I'm seeing.

The default filament diameter for my printer is 1.75mm.  The actual "measured" diameter is 1.72mm.  When I enter 1.72 in the box in Machine Settings it will accept it but it _does not_ change the Printer Settings box and it is the number in the Printer Settings "Diameter" box that appears to be used for the calculation of the E values.
Even though the Machine Settings "Compatible Diameter" box accepted my non-standard number, it doesn't get used until I manually change the Printer Settings Diameter box. 
Printer Settings is not loaded by default and there is no indication that a user's "custom" diameter _is not_ going to be used in the calculations.
If I change the Machine Settings "Compatible Diameter" to <=2.50 then the Printer Settings "Diameter" will be 1.75.
If I change the Machine Settings "Compatible Diameter" to >2.51 but <3.50 then Printer Settings "Diameter" will be 2.85.
If I change the Machine Settings "Compatible Diameter" to >=3.50 then Printer Settings "Diameter" will be 1.75.
So somewhere in the code is an "illogic" statement that limits the number passed along (to Printer Settings) to either 1.75 or 2.85.  That is the problem that the OP ran across when attempting to enter 12.0 in the box in Machine Settings.  The box accepted it but Cura didn't use it.

The box in Machine Settings is clearly labeled as "Compatible Material Diameter" so maybe it should only accept 1.75 or 2.85.  The ToolTip could be adjusted to advise a user that if they need to enter a non-standard number then they need to load the Printer Settings plugin and enter the number there.

In reverse - whatever number I enter in the Printer Settings Diameter box is sent to the Machine Settings Compatible Diameter box as written.  I enter 1.72 in the Printer Settings and the Machine Settings updates to 1.72 when it should probably update to 1.75 as the closest "Compatible" diameter.

A couple of thoughts:

1. The "Machine Settings Compatible Diameter" box can be limited to entries of standard filament diameters.  Maybe a dropdown list.  The Tooltip for the setting should explain that "Custom" numbers require that the Printer Settings plugin be loaded from the MarketPlace and the custom diameter entered in the box there.
     a. The Machine Settings can update the Printer Settings Diameter box to either 1.75 or 2.85 as a starting point when a printer is installed.  I assume that is the current behavior.
     b. The Printer Settings "Diameter" should be relabeled to "Actual Diameter" to indicate to a user that this is the number used in the calculations.
     c.  When custom numbers are entered in Printer Settings then the "Compatible Diameter" box should update using something similar to the current logic that diameters of <=2.5 become 1.75 and numbers >2.5 become 2.85.  That would make it consistent with only accepting "standard" diameters.

2. An alternative would be to relabel the box in Machine Settings to simply "Filament Diameter" and pass whatever is entered to the box in Printer Settings.  Essentially, just eliminate (what I assume to be) the "IF" statement in the code that limits what is passed to Printer Settings.

If a user changes what they see as the "Diameter" then they will assume (as I did and as the OP did) that they changed the diameter.  The fact that the box says "Compatible" gets missed as the box will accept any number and then the custom number doesn't get passed along to the "real actual" diameter setting box.

Joeydelarago, 2022-12-12:
Good Monday Morning, 

I had another look at this. I've sliced the same file twice. Once with a 1.75mm diameter material and once with 10.0mm diameter. Both were set using only the MachineSettings menu. I see an increase in E values, proportional to the increase in material diameter.

I'm pretty sure that any setting passed in Machine Settings gets resolved and passed to the engine as I have described in the above comment, however the settings are quite complicated so I could be wrong. If you can give me a reproducable example of the issue I will add it to our backlog.

Left "LargeMaterialDiameter.gcode", Right "NormalMaterialDiamter.gcode"

[image]

  Thanks

fieldOfView, 2022-12-12:
> I've sliced the same file twice. Once with a 1.75mm diameter material and once with 10.0mm diameter. Both were set using only the MachineSettings menu. 

I'm pretty sure Cura selected a different material for you when you changed the compatible material diameter from 1.75 to 10.0. Because that is all that setting in the Machine Setting should be used for in Cura; select what materials are available for you. The 1.75 mm setting probably got you a 1.75mm variation of PLA. But since Cura does not have any 10mm materials, it would have given the printer the choice of NO materials, thus defaulting back to the standard value of the material diameter being sent to CuraEngine, which is 2.85.

Joeydelarago, 2022-12-12:
@fieldOfView I created a material with a diameter of 10.0mm for this test. I should have included this in the message, my bad.
You can see that the E commands are around two orders of magnitude larger, which is what I would expect from the difference in area between 1.75mm and 10mm diameter circles

fieldOfView, 2022-12-12:
Ok, I understand what you were trying to say. This is just very confusing:

> I'm pretty sure that any setting passed in Machine Settings gets resolved and passed to the engine as I have described in the above comment, 

As a matter of fact, what you set in the Machine Settings for the diameter is _not_ passed to the engine (and not supposed to either). It is _only_ used as a way to filter the available materials. Changing the value may select a different material or profile, and any value set for the material diameter _in that material or profile_ will be passed to the engine.

GregValiant, 2022-12-12:
I'm playing around with this.  The main thing seems to be the "Extruder Definition".  The "Quality", "Nozzle Size", and "Settings Profile" would also seem to be involved.

If there is an Extruder Definition with "material_diameter": { "default_value": 10.0 } and I set Compatible Filament Diameter to "10.0" in the machine settings then it works.  Printer Settings will update to 10.0.
If I set Compatible Filament Diameter to 15.0 then the Printer Settings box defaults to 1.75 (because it is  > 3.50 as I noted earlier) and that appears to be because there is no "extruder definition" for 15.0.

It appears that "Compatible Filament Diameter" affects the extruder selection.  If there is an extruder with a "material_diameter" equal to "Compatible Filament Diameter" then it gets selected and things work.
If there is no extruder capable of the input "Compatible Filament Diameter" then things default back to 1.75 or 2.85 depending on the logic I noted earlier.

Maybe the "Compatible Filament Diameter" does need to be a drop down list populated by all the "material_diameter" numbers of the available extruders.   The tooltip could say that if further tweaking needed to be done it would need to be done in the "Printer Settings" renamed "Actual Filament Diameter" box.
That would eliminate the "IF" statement that passes along the <2.5, >3.5 logic.  The user could only pick from the "material_diameters" of the available extruders.  

Or something.  I'm with @fieldOfView ...my head hurts from all of this.