Ultimaker/Cura #13901: Flow settings influence the line width in the slice preview

Diluvian2, 2022-11-25:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

(Custom) Ender 5 Pro

### Reproduction steps

1. set flow lower than 100% (e.g. for walls)

### Actual results

1. Flow settings 100%, Line Width 0.7mm -> The calculated wall width in the preview of the sliced result is calculated as 0.7mm
[image]

2. Flow settings 86%, Line Width setting still 0.7mm -> The calculated wall with in the preview is now 0.6mm
[image]

### Expected results

After the reduction of flow I expect that curas algorithm still expects a line width of 0.7mm but the actual 3d printer reduces the flow to actually reach these 0.7mm (and not the 0.8mm width caused by overextrusion)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Simple cube.
[cube.zip](https://github.com/Ultimaker/Cura/files/10095034/cube.zip)

GregValiant, 2022-11-25:
Thanks for the report.
My analysis macro in Excel comes up with .50 line width for the skirt and all Outer-Wall line widths are .60.
86% of .70 is .602 so Cura would appear to be right on the money.  Your Skirt Line Width is 0.50 at 100% flow so that is exactly right as well.
The printer uses the numbers that Cura provided in the Gcode.  It does not adlib.  If you were to alter the global flow rate on the printer (usually located in the Tune menu) then you could make it 114% and pretty much be back at a .70 line width.

If you have calibrated the E-steps on the printer, then you are done.  If you are trying to further calibrate the flow with a single wall cube I would advise you to throw it away.  When the "Volume of Filament In = Volume of Extrusion Out" then your flow is 100%.  There is nothing in there about measuring walls.
(PI)r² x Length of filament = LayerHeight x LineWidth x Length of Extrusion.

The 0.8 line width you may be experiencing at 100% is unlikely to be caused by any over-extrusion.  It might be caused because the extrusion shape is not rectangular as it is unconstrained on both sides.  That will make an oval of the same area as a rectangle and so it will be wider.  The best way to check the flow is on a large skin and under a microscope.  A decent magnifying glass will work.

MariMakes, 2022-12-08:
Hey @Diluvian2,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍 Also, Thanks @GregValiant for your wonderful analysis 🎉 

@Diluvian2, I'm super curious 🤔 . 
Can you share what you are trying to achieve? 
Are you just trying to understand how the line width works?
Or are you trying to solve a specific problem by changing the flow settings here? 

Maybe we can help with your original problem instead. 😊

Diluvian2, 2022-12-09:
Hello @GregValiant and @MariMakes 
Sorry for my very late response, I just didn't find any time to continue my 3D printing experience the last weeks :).

@GregValiant I think you got it right that my impression was to kind of calibrate the printer with the FLOW setting without the need to set the E-Steps. I also feared that if CURA recognizes a low flow rate setting (e.g. 50% FLOW) that it will calculate an additional wall needed to reach the expected width of the model being sliced but CURA does exactly what it should.

So to sum it up somehow:
- With E-Steps you calibrate the printer to deliver the correct flow as expected by the slicer
- With Wall Width setting you set the width to be expected by the slicer for which it shall calculate the travel route and the decision how far the walls are printed apart (so to say)
- With Flow Rate you do not change the initial slicer wall width expectations for calculating the travel route but you can alter the flow rate then to dial in the wall width you expect specifically for the available line types (for some you may want kind of "overextrusion" behavior and for some others you may want to decide to "underextrude" (maybe because the wall width of the final print is to far away from the expectation of cura)
- Additional option is the horizontal expansion setting which is also helpful to further raise the accuracy of the outer surface

Though what may be a nice feature would be to somehow calibrate CURA to better display the reality. If the printer E-Steps are correctly set up (correct flow) and CURA wall width setting is 0.7mm (with 100% flow setting) that the slicer can be corrected with the real outcome (e.g. I've measured 0.8mm actual width). That would lead to a different travel route (e.g. further apart from the model surface - same as horizontal expansion does - but the correction would apply everywhere, even inside the model). But this may need deeper analysis in order to be able to decide if that makes sense and if it's at all necessary.

And yes it seems that this may not be tagged as "bug" anymore. Shall I do that or is it sufficent to just close this report?

Thanks for your help!