Ultimaker/Cura #13903: Adding Draft shield to excludable object

mjrider, 2022-11-26:
### Is your feature request related to a problem?

Having objects marked in the gcode so you can cancel a part of a print during a print already saved me  multiple time.

During a print today a draft shield failed, and in such a way that in the end it fused to the print and the print was lost regretfully.

### Describe the solution you'd like

Adding the same markers in the gcode which make it possible for the printer to exclude a object during the rest of the print to the gcode specific for draft shields could have saved me filament and the spageti of the failed draft shield, and maybe even saved the whole print.

### Describe alternatives you've considered

i don't see how this could be done otherwise

### Affected users and/or printers

everybody using cancelable/excludable object

GregValiant, 2022-12-01:
Isn't this a firmware thing?  I think Octoprint has such a feature and if a problem develops with one model on the build plate it can be excluded.  The marker in the file would be something like ";MESH:Big Model.stl(2)".  You can exclude that one and Big Model.stl(1) and Big Model.stl(3) would still print.  My printer by itself can't do such a thing even if it was named ";MESH:DRAFT_SHIELD".

Within the gcode file the draft shield is "TYPE:SKIRT".  So there is already a marker in the gcode.  How would you tell the printer to exclude it?  The firmware would need to actually read the lines but most firmware ignores any line that starts with a semi-colon because it indicates a "comment" rather than a "command".
The Octoprint plugin for Cura was developed by @fieldOfView so maybe he would have a take on this??

mjrider, 2022-12-08:
this is a firmware/octoprint/klipper thing yes, and cura has support for providing the required information for all object so it works, because cura and the firmware are working together on this, cura provides markers and information, and the printing software knows how to handle it

adding the required information requires more than just the TYPE:SKIRT rename, because it needs a begin and end marker on the right part of the gcode and a registration in the begin of the file with minx/miny maxx/maxy and a name.
Because cura is already generating the 'object' draft shield, adding the required entries in the GCODE should be a lot easier that trying to fix it with a postprocessing script

GregValiant, 2022-12-08:
_"... cura provides markers and information, and the printing software knows how to handle it."_
That doesn't work on my Marlin firmware.  All lines starting with a semi-colon are skipped over completely.  Without another piece of software in between (Octoprint/RaspPi for example) it won't work for me.  So any filtering needs to take place before the gcode makes it to the printer because any G or M code that makes it through will be acted on.  The firmware (for the most part) is sufficient to figure the movements but going beyond that then probably not.

closed 2024-11-03 by mjrider