Ultimaker/Cura #13905: lower movement slider not showing and wierd lines showing

pratyushganguly, 2022-11-26:
### Application Version

5.2.1

### Platform

windows 11

### Printer

custom

### Reproduction steps

1. I uninstalled some plugins

### Actual results

1. Not showing the slider which shows the nozzle movement.
2. showing weird thin lines in the layer lines instead of proper layers (screenshots attached)

[image]
[image]

### Expected results

Show lower slider and properly color coded layers

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

kindly help with any settings (if aplicable).
my cura screen is not showing the layers as it used to show, crisp and clean. (probably in previous versions)
tried 3mf as well as stl

GregValiant, 2022-11-26:
Thanks for the report.
Try going to Preferences / Configure Cura and in the Viewport settings make sure "Force Layer View Compatibility Mode" is not checked.
Your computer must have OpenGL 4.0 or later drivers.  If the computer has two video subsystems you may have to get it to use the more capable one with Cura.
This is probably not a Cura bug.  I'll leave the label on so you can respond here and let us know if you found the problem.

pratyushganguly, 2022-11-29:
[image]

Thanks @GregValiant for your help, both of my problems are resolved by unchecking "Force Layer View Compatibility Mode".
I have not yet checked OpenGL 4.0.

Thanks again.

closed 2022-11-29 by pratyushganguly

GregValiant, 2022-11-29:
If your preview is showing "sausages" instead of flat lines then you must have OpenGL 4.1 or higher.  If you didn't then Cura would default back to Combability mode without any user input.