Ultimaker/Cura #13906: Pause at height feature causes blobs

SergeantStoned, 2022-11-26:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender Max

### Reproduction steps

1. Loaded my model into the slicer
2. Went into post processing to enable  **Pause at height**
3. Sliced my model 
4. Printed from the SD card with the depicted settings
5. This issue only occured to me while using the **Pause at height** post-processing feature
[image]

### Actual results

[image]
I had a couple of failed prints, because once I resumed the print, the printhead moved over to the print and then extruded exactly the retracted amount (in this case 10mm) straight onto the print and then started to print as inteded.

_So the print failed because of a huge blob in one corner of the print._

Please note that I've tried to remove the depicted blob while it was printing, thus it isn't as large as it was when it initially failed.

### Expected results

The **retraction** setting is somehow added on top of the **extrude amount** and thus makes it useless.
[image]

I expect that the **retraction** setting **does not** interfere with the **extrude amount**.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Project files.zip](https://github.com/Ultimaker/Cura/files/10096426/Project.files.zip)

GregValiant, 2022-11-26:
Thanks for the report.
This behavior has been noted.  The problem is that Pause at Height has the prime line over the print instead of at the park position.
I have a re-do of Pause at Height in the works.  Maybe by 5.3 it will be included.
In the meantime, if you can get by without using the retraction feature of Pause at Height you can get rid of the blob.  After switching filament, push the new color through the nozzle by hand.  After you are happy with the purge, pull the filament back by your "retraction distance" (usually 5 or 6mm) and then restart the print.
I'll mark this as a duplicate of #13906.

Asterchades, 2022-11-27:
Honest question in all this... why do so many people insist on using Pause At Height instead of Filament Change to do, well, a filament change? Is there some functionality missing from the latter which makes it less desirable? Are people simply not aware of its existence? Has someone popular written an incorrect guide on the subject?

I'd be disinclined to change the Pause At Height functionality as its current functionality is consistent to the rest of the print. If you pause for any purpose other than a filament change (eg adding a magnet, filling a print with sand, just pausing while you answer the phone, etc) you would expect that it retracts as specified when it pauses, then automatically de-retracts that same value when it resumes. This is how _every other retraction in the print_ behaves by default (there is only one Retraction Distance setting), so changing it for one circumstance creates a weird inconsistency.

Also like any other retraction, if you were to manually feed the material through between the retraction and the de-retraction you'd expect a blob to form when it starts again based on your Retraction Distance. The printer has no way of knowing that you've moved the filament, Cura has no reason to suspect you would, so it would de-retract as normal. This exact same thing happens when you effectively de-retract your new material by feeding it right to the nozzle tip during the pause.

If you must use Pause At Height to change filament, perhaps set the Extrude Amount to the _negative_ equivalent of the Retraction value. So long as this is applied before the tool moves back into place, before the de-retraction movement is performed, or combined into the de-retraction distance it should undo the manual filament position change you've done.

GregValiant, 2022-11-27:
Hi @Asterchades .
For me it is personal preference.  I've gotten used to using Pause at Height and it works well for me.  I use it both for inserting nuts, reinforcements, sand (was actually lead shot), as well as for changing colors.

_"So long as this is applied before the tool moves back into place..."_  Ah-ha!!  That is the crux of the problem.  The un-retraction occurs right after the nozzle moves back to the print when the firmware flavor is not Griffon or RepRap. It's a timing thing.

So I moved the un-retraction so it occurs at the park position rather than above the model (if the firmware is not Griffon or RepRap).  In addition, I had a couple of things on my Christmas list and since I was in the code anyway I added them.  I've been using my version for about 6 months and I like it:

- Multi-Command support for both "Gcode before pause" and "Gcode after pause".  It requires a comma as a delimiter.
- An option to use M104 for the resume temperature because it irked me to have to wait the 10 seconds for resume with M109 even when the print temperature was equal to my standby temperature.

So beyond "when" the un-retraction occurs (in Marlin firmware) I didn't make any changes.  Maybe the powers that be at UM will throw it out but I'll continue to use it with my changes.

In regards to M600 - I have no idea how Creality has configured it in their firmware on my printer.  If it's anything like other aspects of their firmware I'll stick to Pause at Height.

Asterchades, 2022-11-27:
> In regards to M600 - I have no idea how Creality has configured it in their firmware on my printer. If it's anything like other aspects of their firmware I'll stick to Pause at Height.

So very, very true. Honestly I'm surprised Pause At Height works for so many people. I've seen reports of Creality firmware actually not waiting for user input for a pause - it just holds for about 6 seconds, then continues on its merry way.

GregValiant, 2022-11-27:
_"I've seen reports of Creality firmware actually not waiting for user input for a pause - it just holds for about 6 seconds, then continues on its merry way."_
When Creality went to the 4.2.x boards they also went to a TFT style LCD and the two didn't play well together.  All the commands that send messages to the LCD (M0, M1, M117, maybe M600) didn't work anymore as the printer would just ignore them.  The pause was only because the next line was the M109 line which much of the time (not always) causes a 10 second delay to the restart.
I have a couple of workarounds for the Creality issue but they involve using Search and Replace because it supports multi-line insertions.  That's actually why I added multi-line to Pause at Height.  It's still a workaround but would only require the single post processor and multiple pauses would work whereas with Search and Replace things got complicated for users real fast if there was more than a single pause.
Your point about the printer losing track of the filament when it's changed is well taken.  Totally eliminating any blob is dependent on how far the user inserts the new filament.
But that prime move after the Pause was (in my opinion anyway) wrong as it always caused a fair sized blob.  
G1 F9000 X94.6 Y115.37 ;Move the resume position
G1 F300 Z4.4 ; move back down to resume height
G1 F1500 E5
If the "Retraction" setting was used then the prime line was always in the gcode.  If the user entered "30" for the retraction distance and then left the "Extruder" box "0" then the gcode would be:
G1 F9000 X94.6 Y115.37
G1 F300 Z4.4 ; move back down to resume height
G1 F1500 E30
That's a pretty big blob and was unexpected because the "extrude" amount was entered as 0.

At any rate, here is what I've proposed.
[GV_PauseAtHeight.zip](https://github.com/Ultimaker/Cura/files/10098635/GV_PauseAtHeight.zip)

SergeantStoned, 2022-12-02:
@GregValiant You're welcome. That's exactly what I did, I've just set retraction to zero and pushed the filament manually trough the nozzle, then waited a bit, cleaned the nozzle and then it continued without creating blobs. Worked flawlessly.

Thank you for your work! Much appreciated.

SergeantStoned, 2022-12-02:
> Honest question in all this... why do so many people insist on using Pause At Height instead of Filament Change to do, well, a filament change? Is there some functionality missing from the latter [...]

@Asterchades I was looking for dual color printing tips and someone mentioned that pause at height works great for dual color prints with a single extruder. I don't know if the filament change script keeps the printing temp up but that was a factor for me as I wanted to switch filaments as quickly as possible.